respeaker/seeed-voicecard #91: /etc/asound.conf issue

jy6666666, 2018-07-24:
Hi all,

i have purchased ReSpeaker_2_Mics_Pi_HAT for raspberry pi 3, i also have a speaker connected with ReSpeaker_2_Mics_Pi_HAT. after i pack ReSpeaker_2_Mics_Pi_HAT and raspberry pi 3 and install the package, i open the **/etc/asound.conf** in command line, i got the following:
**pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}
pcm.playback {
    type plug
    slave.pcm _"dmixed"_
}
pcm.capture {
    type plug
    slave.pcm _"array"_
}
pcm.dmixed {
    type dmix
    slave.pcm "hw:1,0"
    ipc_key 555555
}
pcm.array {
    type dsnoop
    slave {
        pcm "hw:1,0"
        channels 2
    }
    ipc_key 666666
}**

with the above setting, i cannot get any sound out, however, if i replaced the "dmixed" and "array" to "hw:1,0", that would work perfectly. BUT every time i plug in my raspberry pi 3 again, /etc/asound.conf shows above setting again (i.e."hw:1,0" has been changed back to "dmixed" and "array").

any ideas on that??

thanks

jy6666666, 2018-07-25:
@Pillar1989 thank you Pillar, is there any other big difference between dev branch and master branch. and if it is possible, do you have any approximate timeline plan for updating this issue in master branch.

ragards

Pillar1989, 2018-07-26:
Thank you for you pay close attention to seeed-voicecard repo, dev branch have been merged to master branch.

jy6666666, 2018-07-26:
@Pillar1989  thank you Pillar. The problem has been solved once I reinstall.

closed 2018-08-06 by KillingJacky