respeaker/seeed-voicecard #95: asoundrc configuration and arecord

onsky-hieu, 2018-08-29:
Hi,
I have purchased the 6mic array kit and overall I am happy to use it. However, I have some unclear points and hopefully you can shed some light on my concern.

I looked at asound_6mic.conf file for 6-mic circular array kit and under the capture part
```
pcm.ac108 {
    type plug
    slave {
        rate 48000
        format S32_LE
        pcm "hw:seeed8micvoicec"
    }
}
```

Obviously, it does not mention about the number of channels within pcm.ac108. However, in README.md, respeaker recommended to explicitly use **-c 8**:
```
#for 6-Mics Circular Array Kit and 4-Mics Linear Array Kit
#It will capture sound on AC108 and save as a.wav
arecord -Dac108 -f S32_LE -r 16000 -c 8 a.wav
#Take care of that the captured mic audio is on the first 6 channels
```

Could you please explain more about this?

Pillar1989, 2018-08-29:
There are a total of 8 channels of data. The first 6 channels are the data of microphone, and the second 2 are the data of playback record channel. It is necessary to specify channel 8 to get the data of playback record channel, You also use less than 8 channels, but you lose data from the playback record channel or other channels

onsky-hieu, 2018-08-29:
Thank you for the quick reply.
Yes I understand there are 8 channels in total. However, I do not know why this information is not listed out on alsa configuration. I expect something like "channels 8" within it.

Pillar1989, 2018-08-29:
If "channel 9" is specified, if less than 8 channels of data are recorded, the result will be mixed.   that's not right.