respeaker/seeed-voicecard #101: Audio Issues ReSpeaker 6-Mic Circular Array

bpmct, 2018-09-17:
Hi. I am having issues with both mic & headphones for the ReSpeaker 6 Mic Array on a Raspberry Pi 3B. After running the install script from the wiki, everything seemed to be working fine, except all channels are silent and I cannot get headphone output.

I made sure I have I2C enabled, and have tried with both SPI on and off.

`$ arecord -L

null
    Discard all samples (playback) or generate zero samples (capture)
ac108
dmixer
ac101
default
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Hardware device with all software conversions`

`aplay -L

null
    Discard all samples (playback) or generate zero samples (capture)
ac108
dmixer
ac101
default
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Hardware device with all software conversions`

**Some screenshots with all recording devices:**

[image]

[image]

[image]

I also made sure that all inputs/outputs are at 100% on `alsamixer`, and made sure `1-wire` is disabled in raspi-config. 

Is there anything I'm missing? Thank you!

turmary, 2018-09-18:
hi @bpmct,
That's may caused by bad contact between Mic-Array board(6 mics on it) and the Hat-Board (inserted into RPi).

bpmct, 2018-09-18:
Hi @turmary! Thanks for your response. I just tried reconnecting the ribbon cable several times, and no such luck.

[image]

I am able to get LEDs working, and I hear a crackle in the headphones whenever I try to play back the (empty) recordings in Audacity. Is there anything else I can try?

turmary, 2018-09-19:
OK, paste your system version here, getted by below commands.
uname -a
lsb_release -a
cat /boot/issue.txt

We 'll test ourselves.

bpmct, 2018-09-19:
@turmary Thanks! Here is the information:

```
**$ uname -a**
Linux vaha1 4.14.69-v7+ #1141 SMP Mon Sep 10 15:26:29 BST 2018 armv7l GNU/Linux
```

```
**$ lsb_release -a**
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

```
**$ cat /boot/issue.txt**
Raspberry Pi reference 2018-06-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0d34fb94398a4bd79711f21d6a0b5d35fe5f4a76, stage5
```
Thanks!

turmary, 2018-09-25:
Hi @bpmct 
We tested the driver on a new Raspbian image 2018-06-27.
The record & play is worked well.
Please download a new image from raspberrypi.org , burn a boot SD Card, then install this driver directly to eliminate the effect of other software/drivers.

closed 2018-10-12 by turmary

bpmct, 2018-10-13:
Hi @turmary 

So sorry for the late response. Just got the time to try everything out.

I installed a fresh OS and tried with the drivers and same exact result.

I even tried using the image with the drivers pre-installed. Same exact results, even from the CLI. I also tried re-seating the cable multiple times.

Attached is the 10s file that I recorded via the command line. It has no sound at all. What else can I do?

[d.wav.tar.gz](https://github.com/respeaker/seeed-voicecard/files/2475930/d.wav.tar.gz)