respeaker/seeed-voicecard #102: [Respeaker4] - Statics, very bad statics

Psychokiller1888, 2018-09-29:
Using a respeaker 4 on a raspberry pi 3b+, recording using your wiki test settings. Changed rate to 42k to see if any difference, it's the same, the statics just being higher tuned

How do I fix this?

[respeaker4.zip](https://github.com/respeaker/seeed-voicecard/files/2430545/respeaker4.zip)

Psychokiller1888, 2018-09-30:
And yes, alsamixer is set up

[image]

Psychokiller1888, 2018-10-03:
This is PGA set to 0....

[test.zip](https://github.com/respeaker/seeed-voicecard/files/2442961/test.zip)

So it doesn't work

Psychokiller1888, 2018-10-03:
And this is the result of these settings:

[image]

You don't even hear me speaking....

[test.zip](https://github.com/respeaker/seeed-voicecard/files/2442973/test.zip)

xiongyihui, 2018-10-07:
How did you record?

Try `arecord -f S16_LE -r 48000 -c 4 -v audio.wav` to record 4 channels audio, and use audacity to check each channel to verify if any microphone is broken.

To check if the ADC is broken or not, we can use a tape to seal the microphone's hole, and check if there are any difference.

Psychokiller1888, 2018-10-08:
Using your settings

[audio.zip](https://github.com/respeaker/seeed-voicecard/files/2457268/audio.zip)

Psychokiller1888, 2018-10-08:
[image]

xyu6, 2018-10-10:
Hi there, 

please try to reduce the capture vol to 50% and see if any issue. If the issue is still there, please drop a mail to techsupport@seeed.cc. thanks. 

best rgds
Bill

Psychokiller1888, 2018-10-10:
Same, but at least I did not bleed from my ears when aplaying this time, as it's lower....

[audio.zip](https://github.com/respeaker/seeed-voicecard/files/2463957/audio.zip)

[image]

[image]

Pillar1989, 2018-10-10:
It looks like there's something wrong with your hardware. It can't be the sound of our microphone array.

Psychokiller1888, 2018-10-10:
Yes, it looks, or sounds like there's something wrong, but what?

Psychokiller1888, 2018-10-12:
So what, I wrote an email with no answers and as a solution you saying it's not your microphone???

xiongyihui, 2018-10-13:
@xyu6

Psychokiller1888, 2018-10-21:
The support for your product is really really hopeless

xyu6, 2018-10-22:
Hi there, what about your email address? I will follow up with you directly and then work on the return process. Sorry, we do not receive your mail on techsupport@seeed.cc mail system. It may be the mail delivery issue. thanks for understanding. Or you can drop bill.yu@seeed.cc directly. thanks.

closed 2018-11-05 by Pillar1989

daenuprobst, 2019-02-03:
I have similar problems, there is quite some static when recording (that or the recording volume is really low). Did you manage to fix the problem @Psychokiller1888

Psychokiller1888, 2019-02-10:
Throw your respeaker to trash and get something else. There's nothing to be done and seeed will not help you in fixing the issue

ejay-ibm, 2019-03-11:
Got the same issue . I have better sound with a 5 euro simple USB mic . 
I try all tunning setting i'm still getting trashy noise that prevent clear speech . 
Sad.