respeaker/seeed-voicecard #105: Audio not working for Respeaker 6-mic on Raspberry Pi 3

bpmct, 2018-10-13:
Opening up a new issue because I took too long to respond on issue 101, and it was closed.

I'm unable to get microphone audio working after a few fresh installs. You can see screenshots in the old issue.

I installed a fresh OS and tried with the drivers and same exact result.

I even tried using the image with the drivers pre-installed. Same exact results, even from the CLI. I also tried re-seating the cable multiple times.

Attached is the 10s file that I recorded via the command line. It has no sound at all. What else can I do?

[d.wav.tar.gz](https://github.com/respeaker/seeed-voicecard/files/2475930/d.wav.tar.gz)

_Originally posted by @bpmct in https://github.com/respeaker/seeed-voicecard/issues/101#issuecomment-429582672_

Pillar1989, 2018-10-15:
Hi, Can you show me your alsamixer configure ?

bpmct, 2018-10-15:
Hi @Pillar1989. Thanks for the response:

Attached was my current alsamixer config:

[image]
[image]
[image]
[image]

I also tried with all values at 90% and I am still having no luck recording with audio.

Pillar1989, 2018-10-16:
in https://github.com/respeaker/seeed-voicecard/issues/101#issuecomment-429582672_  There was a voice in your previous comments, but the voice was not correct. Now there is no voice at all. Can you describe your problem clearly if you want our help?

[image]

bpmct, 2018-10-16:
@Pillar1989 My issue is that my microphone is not recording any audio. It just returns an empty audio file. What do you mean there was a voice in my previous comments?

Pillar1989, 2018-10-17:
Yes

bpmct, 2018-10-18:
@Pillar1989 I don't understand sorry. Could you please explain?

Pillar1989, 2018-11-05:
@bpmct Will close this issue, please open a new issue if you have further questions.

closed 2018-11-05 by Pillar1989