respeaker/seeed-voicecard #106: No Dsnoop on B/B+ with 4-Mic Raspberry Pi. Cannot record from multiple.

willondubs, 2018-10-14:
Hi, I thought I was having this problem because I bought a rPi3B instead of a B+. I have a B+ now and am having the same problem. 

My project has 3 different processes that need the mic. This is allowed by alsa and the dsnoop plugin. I installed per the instructions and uncommented the dmix and dsnoop sections in /etc/asound.conf. However, I'm unable to get this to work. Here's my arecord -l output:

$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec0-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

The instructions say my output should be different and show dmix/dsnoop ability but it doesn't. I'm hoping someone can help me get this working. Thanks.

willondubs, 2018-10-15:
Greetings, following up that I have dsnoop working now by modifying the asound.conf. I'm now hoping to get some assistance with further edits to my alsa config so I can do an aplay or arecord without errors or warnings. Here's my asound.conf:
```
$ cat /etc/asound.conf
pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "ac108"
}

pcm.playback {
     type dmix
    slave.pcm "hw:ALSA"
     ipc_key 555555 

}

pcm.ac108 {
    type dsnoop
    ipc_key 666666
    slave {
           pcm "hw:seeed4micvoicec"
           rate 48000
    }
}
```

aplay output:
```
$ aplay test.wav
Playing WAVE 'test.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
aplay: set_params:1299: Sample format non available
Available formats:
- S16_LE
```

arecord output:
```
$ arecord test.wav
Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
arecord: set_params:1299: Sample format non available
Available formats:
- S32_LE
$ arecord -f S32_LE -c 4 test.wav
Recording WAVE 'test.wav' : Signed 32 bit Little Endian, Rate 8000 Hz, Channels 4
Warning: rate is not accurate (requested = 8000Hz, got = 48000Hz)
         please, try the plug plugin (-Dplug:ac108)
```
So if you could help me shore up my asound.conf to remove those messages, I'd greatly appreciate it!

Pillar1989, 2018-10-15:
About 4mic-hat with multiple you can refer to the https://github.com/respeaker/seeed-voicecard/commits/master/asound_4mic.conf history . In the early days we also used the relevant configuration. I hope this helps

willondubs, 2018-10-17:
Thank you @Pillar1989. Do you think all my issues are with alsa configuration?

For example, I'm currently unable to record at rate 8000. Can respeaker record at rate 8000? I'm very familiar with alsa configuration, thanks. Here's my current asound.conf:
```
pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "ac108"
}

pcm.playback {
    type plug
    slave.pcm "hw:0,0"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:0,0"
    ipc_key 555555 
}

pcm.ac108 {
    type hw
    card 1
    device 0
    channels 4
}
```
Output of arecord test.wav:
```
$ arecord test.wav
Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
arecord: set_params:1299: Sample format non available
Available formats:
- S32_LE
```
Only record when I run: arecord -f S32_LE -c 4 test.wav

I'm trying to run project: http://www.jezra.net/projects/blather But I think it wants rate 8000.

closed 2018-10-19 by willondubs