respeaker/seeed-voicecard #107: arecord -L not right

zhaoforever, 2018-10-19:
root@raspberrypi:/home/pi# arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
ac108
dmixer
ac101
root@raspberrypi:/home/pi#

root@raspberrypi:/home/pi# cat /var/log/seeed-voicecard.log
+ dtparam i2c_arm=on
+ modprobe i2c-dev
+ dtparam spi=on
+ sleep 1
++ i2cdetect -y 1 0x1a 0x1a
++ grep 1a
++ awk '{print $2}'
+ is_1a=1a
++ i2cdetect -y 1 0x35 0x35
++ grep 35
++ awk '{print $2}'
+ is_35=35
++ i2cdetect -y 1 0x3b 0x3b
++ grep 3b
++ awk '{print $2}'
+ is_3b=3b
+ rm /etc/asound.conf
rm: cannot remove '/etc/asound.conf': No such file or directory
+ rm /var/lib/alsa/asound.state
+ '[' x1a == x1a ']'
+ '[' x35 == x ']'
+ '[' x3b == x3b ']'
+ '[' x35 == x ']'
+ '[' x3b == x3b ']'
+ '[' x35 == x35 ']'
+ echo 'install 6mic'
install 6mic
+ dtoverlay seeed-8mic-voicecard
+ sleep 1
+ rm /etc/asound.conf
rm: cannot remove '/etc/asound.conf': No such file or directory
+ rm /var/lib/alsa/asound.state
rm: cannot remove '/var/lib/alsa/asound.state': No such file or directory
+ echo 'create 6mic asound configure file'
create 6mic asound configure file
+ ln -s /etc/voicecard/asound_6mic.conf /etc/asound.conf
+ echo 'create 6mic asound status file'
create 6mic asound status file
+ ln -s /etc/voicecard/ac108_6mic.state /var/lib/alsa/asound.state
+ alsactl restore
+ amixer cset numid=3 1
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
  : values=1

Pillar1989, 2018-10-19:
Send me the result of the cmd  ` i2cdetect -y 1`

Pillar1989, 2018-10-19:
This is because you have a local network problem that makes downloading impossible.Please select your current area apt source to download again.

https://www.raspbian.org/RaspbianMirrors

zhaoforever, 2018-10-19:
Do you mean I need to use the original mirrors instead of the Tsinghua mirror?

zhaoforever1@163.com
 
From: Baozhu Zuo
Date: 2018-10-19 17:27
To: respeaker/seeed-voicecard
CC: zhaoforever; Author
Subject: Re: [respeaker/seeed-voicecard] arecord -L not right (#107)
This is because you have a local network problem that makes downloading impossible.Please select your current area apt source to download again.
https://www.raspbian.org/RaspbianMirrors

closed 2018-10-22 by zhaoforever

PatrickYung, 2018-11-04:
@zhaoforever 
May I know how do you settle the problem finally?

I think I got the same problem as yours that failed to install raspberrypi-kernel-headers.

pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Hit https://mirrors.tuna.tsinghua.edu.cn jessie InRelease
Hit https://deb.nodesource.com jessie InRelease
Hit https://mirrors.tuna.tsinghua.edu.cn jessie/main Sources
Hit https://deb.nodesource.com jessie/main Sources
Hit https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Sources
Hit https://deb.nodesource.com jessie/main armhf Packages
Hit https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Sources
Hit https://mirrors.tuna.tsinghua.edu.cn jessie/main armhf Packages
Hit https://mirrors.tuna.tsinghua.edu.cn jessie/contrib armhf Packages
Hit https://mirrors.tuna.tsinghua.edu.cn jessie/non-free armhf Packages
Get:1 https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Translation-en [169 B]
Get:2 https://mirrors.tuna.tsinghua.edu.cn jessie/main Translation-en [169 B]
Get:3 https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Translation-en [169 B]
Get:4 https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Translation-en [169 B]
Get:5 https://deb.nodesource.com jessie/main Translation-en [162 B]
Get:6 https://mirrors.tuna.tsinghua.edu.cn jessie/main Translation-en [169 B]
Get:7 https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Translation-en [169 B]
Get:8 https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Translation-en [169 B]
Get:9 https://mirrors.tuna.tsinghua.edu.cn jessie/main Translation-en [169 B]
Get:10 https://deb.nodesource.com jessie/main Translation-en [162 B]
Get:11 https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Translation-en [169 B]
Get:12 https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Translation-en [169 B]
Get:13 https://mirrors.tuna.tsinghua.edu.cn jessie/main Translation-en [169 B]
Get:14 https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Translation-en [169 B]
Get:15 https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Translation-en [169 B]
Ign https://mirrors.tuna.tsinghua.edu.cn jessie/contrib Translation-en
Get:16 https://mirrors.tuna.tsinghua.edu.cn jessie/main Translation-en [169 B]
Ign https://mirrors.tuna.tsinghua.edu.cn jessie/main Translation-en
Get:17 https://deb.nodesource.com jessie/main Translation-en [162 B]
Get:18 https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Translation-en [169 B]
Ign https://mirrors.tuna.tsinghua.edu.cn jessie/non-free Translation-en
Get:19 https://deb.nodesource.com jessie/main Translation-en [162 B]
Get:20 https://deb.nodesource.com jessie/main Translation-en [162 B]
Ign https://deb.nodesource.com jessie/main Translation-en
Reading package lists... Done
Building dependency tree
Reading state information... Done
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
**E: Unable to locate package raspberrypi-kernel-headers**
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version.
git is already the newest version.
libasound2-plugins is already the newest version.
i2c-tools is already the newest version.
The following packages were automatically installed and are no longer required:
  libc-ares2 libv8-3.14.5
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....(bad exit status: 2)
make KERNELRELEASE=3.6-trunk-rpi -C /lib/modules/3.6-trunk-rpi/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.6-trunk-rpi (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
Error! echo
Your kernel headers for kernel 4.4.50+ cannot be found at
/lib/modules/4.4.50+/build or /lib/modules/4.4.50+/source.
Error! echo
Your kernel headers for kernel 4.4.50-v7+ cannot be found at
/lib/modules/4.4.50-v7+/build or /lib/modules/4.4.50-v7+/source.
Error! echo
Your kernel headers for kernel 4.9.19+ cannot be found at
/lib/modules/4.9.19+/build or /lib/modules/4.9.19+/source.
Error! echo
Your kernel headers for kernel 4.9.19-v7+ cannot be found at
/lib/modules/4.9.19-v7+/build or /lib/modules/4.9.19-v7+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) b806056] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink from /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service to /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------