respeaker/seeed-voicecard #108: install error

zlotus, 2018-10-24:
更新了最新的raspbian和固件，运行`install.sh`：
```bash
pi@raspberrypi:~/seeed/seeed-voicecard $ sudo ./install.sh
命中:1 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch InRelease
命中:2 http://archive.raspberrypi.org/debian stretch InRelease
正在读取软件包列表... 完成
正在分析软件包的依赖关系树
正在读取状态信息... 完成
所有软件包均为最新。
正在读取软件包列表... 完成
正在分析软件包的依赖关系树
正在读取状态信息... 完成
raspberrypi-kernel 已经是最新版 (1.20180924-1)。
raspberrypi-kernel-headers 已经是最新版 (1.20180924-1)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。
正在读取软件包列表... 完成
正在分析软件包的依赖关系树
正在读取状态信息... 完成
dkms 已经是最新版 (2.3-2)。
git 已经是最新版 (1:2.11.0-3+deb9u4)。
i2c-tools 已经是最新版 (3.1.2-3)。
libasound2-plugins 已经是最新版 (1.1.1-1)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.14.71+ (armv7l)
-------------------------------------

Status: This module version was INACTIVE for this kernel.
depmod....

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.14.71-v7+ (armv7l)
-------------------------------------

Status: This module version was INACTIVE for this kernel.
depmod....

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.14.71-v7+ (armv7l)
-------------------------------------

Status: This module version was INACTIVE for this kernel.
depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.71+ -C /lib/modules/4.14.71+/build M=/var/lib/dkms/seeed-voicecard/0.3/build........
cleaning build area....

DKMS: build completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.71-v7+ -C /lib/modules/4.14.71-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area...

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.71-v7+ -C /lib/modules/4.14.71-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area...

DKMS: build completed.
Error! echo
Your kernel headers for kernel 4.14.78+ cannot be found at
/lib/modules/4.14.78+/build or /lib/modules/4.14.78+/source.
Error! echo
Your kernel headers for kernel 4.14.78-v7+ cannot be found at
/lib/modules/4.14.78-v7+/build or /lib/modules/4.14.78-v7+/source.
mkdir: 无法创建目录"/etc/voicecard": 文件已存在
git init
重新初始化已存在的 Git 仓库于 /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
位于分支 master
无文件要提交，干净的工作区
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

按照[ReSpeaker 6-Mic Circular Array Pi HAT](http://wiki.seeedstudio.com/cn/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/#3)测试：
```bash
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
ac108
dmixer
ac101
```
```bash
pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
ac108
dmixer
ac101
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
```
升级系统的时候隐约瞟到系统文件有不少`mv`的情况，不知道是否导致上面的：
```bash
Error! echo
Your kernel headers for kernel 4.14.78+ cannot be found at
/lib/modules/4.14.78+/build or /lib/modules/4.14.78+/source.
```

Pillar1989, 2018-10-25:
https://v2.fangcloud.com/share/7395fd138a1cab496fd4792fe5 这里有已经安装好驱动的固件

zlotus, 2018-10-26:
@Pillar1989 Oh, I see...I installed seeed-voicecard successfully in a brand new raspbian without update the firmware. 
So, just don't run `sudo rpi-update && sudo reboot -h now`.

closed 2018-10-26 by zlotus