respeaker/seeed-voicecard #110: Respeaker micarray v2.0 not recognized on Raspberry Pi 3

amitprakash, 2018-10-27:
I see the MicArray power up and detect voice direction based on LEDs lighting up. But The Pi will not recognize the device.

root@raspi3:~# lsusb
Bus 001 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120 for Business
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

------------------------------
root@raspi3:~# arecord -l
**** List of CAPTURE Hardware Devices ****
---------------------------------
root@raspi3:~/usb_4_mic_array# sudo python dfu.py --download 6_channels_firmware.bin
Traceback (most recent call last):
  File "dfu.py", line 229, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "dfu.py", line 218, in main
    dev = XMOS_DFU()
  File "dfu.py", line 198, in __init__
    super(XMOS_DFU, self).__init__()
  File "dfu.py", line 73, in __init__
    raise ValueError('No DFU device found')
ValueError: No DFU device found

xiongyihui, 2018-10-29:
Was it recognized by your computer? Sometimes usb cable may cause the device unrecognized.

amitprakash, 2018-10-29:
lsusb command did not list the device. arecord -l did not either, as I posted above. I have tried 2 usb cables that I had. Ordered one more fresh USB connector. Will see tomorrow.