respeaker/seeed-voicecard #112: Respeaker and Mycroft dmesg ac10x_update_bits() error

ingoATgithub, 2018-11-04:
Hi,

i'm trying to use the Respeaker 4mic with Mycroft.AI (https://mycroft.ai/) on a raspberry pi 3. The installation of ReSpeaker and Mycroft was straight forward (following the installation instructions), the recording and replaying tests were successful.

```
pi@picroft:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@picroft:~ $ uname -a
Linux picroft 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
```

```
pi@picroft:~ $ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

But when i'm starting mycroft, after some seconds (the mic is working till then properly, i can ask questions and so on) the mic freezes and i get the following error message in dmesg:

```
[   29.880159] ac10x_update_bits() error->[REG-0x30,val-0x00]
[   29.880169] ac10x_update_bits() error->[REG-0x10,val-0x00]
[   29.880176] ac10x_update_bits() error->[REG-0x30,val-0x80]
```

I tried several configurations, installed Respeaker and Mycroft several times (e.g. on latest Raspbian Stretch Lite); this now is a prepared Mycroft image based on Raspbian), but every time the same issue.

What am i doing wrong ? Or can you give me a hint what is going wrong ? Using another mic via USB soundcard instead of respeaker, mycroft is running well.

[(https://github.com/respeaker/seeed-voicecard/issues/20)] did not work for me.

Many thanx !

jasonyork, 2018-11-05:
I'm seeing a similar problem with on a Pi 3 B+, latest Raspbian with my 6-mic circular array kit.

I'm able to record and playback using `aplay` /  `arecord`.  However when trying to use snips, the `snips-asr` is constantly crashing with the syslog messages repeating:
```
Nov  5 02:44:41 raspberrypi systemd[1]: snips-asr.service: Main process exited, code=killed, status=4/ILL
Nov  5 02:44:41 raspberrypi systemd[1]: snips-asr.service: Unit entered failed state.
Nov  5 02:44:41 raspberrypi kernel: [11822.544841] ac10x_update_bits() error->[REG-0x30,val-0x00]
Nov  5 02:44:41 raspberrypi kernel: [11822.544866] ac10x_update_bits() error->[REG-0x30,val-0x00]
Nov  5 02:44:41 raspberrypi kernel: [11822.544876] ac10x_update_bits() error->[REG-0x10,val-0x00]
Nov  5 02:44:41 raspberrypi kernel: [11822.544886] ac10x_update_bits() error->[REG-0x10,val-0x00]
```

```
i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- UU -- -- -- -- -- UU -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

turmary, 2018-11-05:
Hi @ingoATgithub @jasonyork.
We 'll try to test by ourselves, please wait some time.

turmary, 2018-11-08:
Hi @ingoATgithub  @jasonyork 
We have tried to fix issue, please reinstall driver, then test your applications again.

ingoATgithub, 2018-11-09:
@turmary 

Thanx, I don't have the ac10x_update_bits() error anymore. But unfortunately the mic still stops responding after some seconds, now showing 

```
[   28.094201] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[  164.150848] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[  223.441746] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[  343.579252] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[  826.635117] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
```

via dmesg.

turmary, 2018-11-10:
@ingoATgithub 
That's maybe cause by a low speed SDCard.
Make sure your are using Class 10 or Higher SDCard.

jasonyork, 2018-11-11:
@turmary - I'm still having a similar problem as @ingoATgithub with the `I2S SYNC error`.  I'm using a U3 SD card, so it should be plenty fast.  I'm going to see if I can get a clean install with just snips running correctly.  Then I will install the respeaker drivers to try and isolate the problem.

turmary, 2018-11-13:
@ingoATgithub  @jasonyork 
Please take care below two limitations:
1.  capture must be start first, or else the capture channels will possibly be disorder.
2.  playback output channels must fill with 8 same channels data or 4 same stereo channels data, or else the speaker or headphone will output nothing possibly.**

I don't know whether Mycroft.AI & snips-asr followed.

closed 2018-11-20 by Pillar1989