respeaker/seeed-voicecard #113: Can a guide for pulseaudio and 2 mic hat be created

dannystaple, 2018-11-11:
I'm yet to find a 100% replicable solid way to link the 2-mic hat with pulse audio (for use with mycroft for example). Could the documentation, or better yet a script be provided to prepare this please?

Pillar1989, 2018-11-13:
Just `do sudo apt install pulseaudio`, I dont think you need more documents.

dannystaple, 2018-11-13:
Thanks for you reply,

That doesn't work - pulse audio needs to be started (it's not on Raspbian
minimal), it needs a configuration to make the respeaker card the default.
Installing it is obviously a necessary step, but it's likely that an
/etc/systemd/system/pulseaudio.service unit file must be created, along
with pulse audio configuration files for the device.
The idea would be to get the user to the point where paplay and parecord
will work directly with the card (and therefore, without much further
fiddling, software like MyCroft).

Danny

qqkl662, 2018-11-16:
hey,guys ,I meet a problem when i debug this device,when i add the "dueros-auth",the command display the   code such as "xdg-open: no method available for opening 'http://127.0.0.1:3000'".I cannot solve it ,

Pillar1989, 2018-11-19:
@qqkl662 I think it's better for you to go straight to the BBS of dueros

closed 2019-01-14 by Pillar1989