respeaker/seeed-voicecard #115: how to mute mic array

Devanshi1510, 2018-11-19:
Hi, I want to mute mic array. I tried this command from pulseaudio:  pactl set-source-mute 1 toggle 
But It didnt disable mic array when I tried to record voice using pactl set-source-mute 1 toggle I was able to record my voice,
While volume control was possible using pactl. I want to mute and unmute mic array like alexa/google home provides.

baorepo, 2018-11-20:
Can you tell me which mic array is used ?

Devanshi15, 2018-11-20:
ReSpeaker mic array V2.0 @baorepo

Pillar1989, 2018-11-20:
I'm so sorry, I didn't find the option which you want in usb_4_mic_array software.
https://github.com/respeaker/usb_4_mic_array/blob/master/tuning.py#L19

But I think "mute mic array" is easy to implement in your application. "mute mic array" means get zero from the sound card. Replace real data with zero

Devanshi15, 2018-11-22:
I implemented it using pulseaudio and it works using this command  - pacmd set-source-mute <index> 1 
Thank you for help!!!

Pillar1989, 2018-11-22:
I will close this issue, the later you still have questions, you can reopen it

closed 2018-11-22 by Pillar1989