respeaker/seeed-voicecard #116: Unable to record and play simultaneously

DamienLaineLinagora, 2018-11-19:
Greetings,

Using pulseaudio, i'm unable to play anysound while recording.

As i refer to provided documentation : "To support multiple applications to record and play simultaneously, try alsa dmix plugins. The used ALSA plugins in the configuration are asym, dmix, dsnoop and plug. Now you can use arecord and audacity to record at the some time. If you can't record or play, try to change the value of ipc_key to another value."

I don't understand that. Can you give me a asound.conf setting or something to use the "dmix plugin" as proposed ?

thanks.

baorepo, 2018-11-20:
Hi,

DamienLaineLinagora, 2018-11-20:
Sure,

I'm using seeed 2 mic Hat on a Raspberry pi 3 with latest Raspian.

I've tried following settings :

With pulse audio and with an overriding .asoundrc 
pcm.pulse {
 type pulse
}
ctl.pulse {
 type pulse
}
pcm.!default {
 type pulse
}
ctl.!default {
 type pulse
}

I'm able to use arecord, aplay with 2 or more audio files being played or recorded. I'm also able to use parecord or paplay yhe same way.

Pulseaudio is configured system wide or as a daemon. Either way it seems to work fine but it doesn't work as intended i'm still unable to  record while i'm playing or unable to play while recording.

By removing completly pulseaudio, using ALSA with the provided configuration (straight out of the box after running the install script in the provided driver https://github.com/respeaker/seeed-voicecard), the  2 mic hat also seems to work well, i can use arecord and aplay but i'm still unable to record and play sounds at the same time.

The documentation, as mentionned in my issue, says that i would have to "try alsa dmix plugins", and i don't understand how to achieve this.

Could you help me with this ?

Thanks

DamienLaineLinagora, 2018-11-20:
Hello :)

-- 
--

Damien LAINE
Engineering Manager - R&D Team @LINAGORA
Mobile: + 33 (0) 648 819 254
Toulouse - France
Research web site :  https://research.linagora.com
LinTO : https://linto.ai
twitter.com/Viral_Art
fr.linkedin.com/in/damienlaine 

Links:
------
[1]
https://github.com/respeaker/seeed-voicecard/issues/116#issuecomment-440106160
[2]
https://github.com/

Pillar1989, 2018-11-20:
@DamienLaineLinagora I promise that in the latest Raspian system, according to the README document, you can achieve  record and play  simultaneously.  README document did not mention such software as pulseaudio, dmix, and so on.

DamienLaineLinagora, 2018-11-20:
The wiki https://github.com/respeaker/seeed-voicecard/wiki says "To support multiple applications to record and play simultaneously, try alsa dmix plugins. The used ALSA plugins in the configuration are asym, dmix, dsnoop and plug. Now you can use arecord and audacity to record at the some time. If you can't record or play, try to change the value of ipc_key to another value."

Straight out of the box with the provided driver, i confirm that on a fresh Raspbian install (using ALSA as far as i stand) i can't record and play at the same time.

Just run a "aplay somefile.wav" after launching a "arecord otherfile.wav", you'll have no output sound as the recorded file will be ok.

I tried to use pulseaudio as it takes full handle over ALSA and this is the way i know to achieve multiple audio clients on the same device.

Pillar1989, 2018-11-22:
@DamienLaineLinagora @Lokhozt 
If record and play simultaneously is not implemented, so that other people do the demo That was all a lie
https://www.hackster.io/idreams/build-your-own-amazon-echo-using-a-rpi-and-respeaker-hat-7f44a0 
### Note: 
https://github.com/respeaker/seeed-voicecard/wiki  is just an overview and does not mention specific sound cards. Please refer to the readme document for the specific sound card.

twestheimer, 2018-12-02:
what was a lie? I don't understand?

-- 
----------------------------------------------------------------------------------------------------------------

   - *To a Buddhist "Self Control" is either a double negative and/or an
   oxymoron *
   -
*I've learned that children and grandparents are natural allies. *
   - *Be loyal to peace, justice, the social good *
   - *Patriotism is your conviction that your country is superior to all
   other countries because you were born in it.      George Bernard Shaw*
   - *"Your assumptions are your windows to the world.  Scrub them off
   every once in a while, or light won't come in."*

----------------------------------------------------------------------------------------------------------------------
Tom Westheimer  *US  phone 603-525-7733 <603-525-7733> (calls cell and
home) *
ทอม  เวสต์ไฮม์เมอร์  Thai phone

*+6685-4466-954*

Pillar1989, 2019-01-14:
I will close this issue, the later you still have questions, you can reopen it

closed 2019-01-14 by Pillar1989