respeaker/seeed-voicecard #117: respeaker 2mic hat audio play is not proper

titusece, 2018-11-19:
Hi,

I am unable to play wav and basic test itself not working in my respeaker board. (2MIC HAT RPI)
I can hear repeaker recorded voice in my PC but RPI0+2MIC respeaker.

Here is the log:

> pi@raspberry:~ $ 
> pi@raspberry:~ $ date
> Mon 19 Nov 19:01:00 UTC 2018
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ aplay --device=hw:1,0 test1.wav  -v^C
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ aplay --device=hw:1,0 test1.wav  -v
> Playing WAVE 'test1.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
> Hardware PCM card 1 'seeed-2mic-voicecard' device 0 subdevice 0
> Its setup is:
>   stream       : PLAYBACK
>   access       : RW_INTERLEAVED
>   format       : S16_LE
>   subformat    : STD
>   channels     : 2
>   rate         : 8000
>   exact rate   : 8000 (8000/1)
>   msbits       : 16
>   buffer_size  : 4000
>   period_size  : 1000
>   period_time  : 125000
>   tstamp_mode  : NONE
>   tstamp_type  : MONOTONIC
>   period_step  : 1
>   avail_min    : 1000
>   period_event : 0
>   start_threshold  : 4000
>   stop_threshold   : 4000
>   silence_threshold: 0
>   silence_size : 0
>   boundary     : 2097152000
>   appl_ptr     : 0
>   hw_ptr       : 0
> ^CAborted by signal Interrupt...
> pi@raspberry:~ $ aplay -l
> **** List of PLAYBACK Hardware Devices ****
> card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
>   Subdevices: 7/7
>   Subdevice #0: subdevice #0
>   Subdevice #1: subdevice #1
>   Subdevice #2: subdevice #2
>   Subdevice #3: subdevice #3
>   Subdevice #4: subdevice #4
>   Subdevice #5: subdevice #5
>   Subdevice #6: subdevice #6
> card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0
> card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ arecord -l
> **** List of CAPTURE Hardware Devices ****
> card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ 
> pi@raspberry:~ $ arecord -f cd -Dhw:1 | aplay -Dhw:1
> Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
> Playing WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
> overrun!!! (at least 1.099 ms long)
> ^CAborted by signal Interrupt...
> Aborted by signal Interrupt...
> stdout: Broken pipe
> pi@raspberry:~ $ 
> pi@raspberry:~ $ dmesg
> [   41.325513] Bluetooth: BNEP socket layer initialized
> [   42.428141] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
> [   60.244342] asoc-simple-card soc:sound: wm8960-hifi <-> 20203000.i2s mapping ok
> [  214.658292] wm8960 1-001a: failed to configure clock
> [  214.658319] wm8960 1-001a: ASoC: Failed to prepare bias: -22
> pi@raspberry:~ $ 
> pi@raspberry:~ $

baorepo, 2018-11-20:
[image]
Is that work for you ?

titusece, 2018-11-20:
Thanks for the reply.
No it didn't work.
aplay --device=hw:1,0 test1.wav -v

I am using raspberry pi zero w board
I have tried to use new raspian OS and it worked that confirms HW works good. Wanted to see whats causing the problem and what configuration to check.

titusece, 2018-11-20:
After disabling the "pigpiod.service" I can hear clear audio now.
What could be reason pigpiod cause this audio issue ?

baorepo, 2018-11-22:
I don't know "pigpiod.service" does.  

I will close this issue, the later you still have questions, you can reopen it

closed 2018-11-22 by Pillar1989