respeaker/seeed-voicecard #119: 2mic hat and pyaudio, playback sampling rate problem

Lokhozt, 2018-11-20:
Hello, 
I've been  trampling on a problem regarding the 2mic hat audio output for a while now.
**The setup (in order):**
**Hardware:** RPi 3B + Respeaker 2-mics Pi HAT + 3V 4Ohm speaker
**OS:** Raspian Stretch Lite (Up to date)
**seed installation:**
```bash
sudo apt-get install git
git clone https://github.com/respeaker/seeed-voicecard.git
cd seed-voicecard
sudo ./install.sh 2mic
pacmd set-default-sink 1
```
The aplay and arecord device lists are: 
```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
**** List of CAPTURE Hardware Devices ****
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
At this point arecord and aplay work.
``` bash
arecord test.wav
> Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
aplay test.wav
> Playing WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono # Speaker works
```
They don't work together though. When arecord is recording, aplay doesn't output anything __except ... if the sampling rate of the audio is 44100Hz__. I've tried other sampling rate (8000, 16000, 32000, 48000) it doesn't seem to work.
On the other hand parecord and paplay work fine together.
The problem is we use for our project several python module using pyaudio both for input and output.
(pyaudio is based on portaudio which make calls to alsa).
We have 2 processes reading the audio input at 16000Hz and 2 processes susceptible to output audio at any given time.
The recording processes are working fine (Hotword detection and Voice Activity detection), however the sound producing processes don't work at all (no error, just no sound or what sounds like a really low energy pinkish noise). Those process are dealing with 16000Hz 16bit 1 channel sounds as well. 
The output works fine if no processes are recording.
As i was investigating the issue I figured out that it only works when the audio is sampled at ... 44100Hz (again).

To resume the situation:
With a process recording from the hat:
1 - play audio at 16000Hz -> No sound
2 - play the same audio resampled at 44100Hz -> sound

***Is there a workaround to this situation ? Am i doing something wrong ?***
A solution could be to resample all our audios to 44100Hz but it is really odd and some audio are dynamically generated (without any control on the output sampling rate) so it would mean add a resampling step during runtime (not so good).

Thank you

PS: This is the same issue as #116 , I've created a new one to display more information and context  and to be more precise about the problem.

baorepo, 2018-11-22:
Let me tell you all the basic knowledge of i2s bus hardware which 2mic hat and Pi direct communication . https://www.sparkfun.com/datasheets/BreakoutBoards/I2SBUS.pdf
record and play use the same clock line. In the case of recording and playing at the same time, the two sampling rate must be consistent to achieve data transfer

Lokhozt, 2018-11-22:
There are consistent. Both recording and playing are set to 16Khz 16bit LE, 1 channel.
However it only work for playing at 44100 Hz despite the fact that the recording run at 16000Hz.