respeaker/seeed-voicecard #121: 4mic hat loud static with stretch-lite 2018-11-13

imsunburnt, 2018-12-22:
I have installed ReSpeaker 4mic hat on raspberry pi 3b+ running stretch-lite-2018-11-13.
When I playback the recording from arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav I get loud static over the very faint recording.
I have tried various releases of stretch-lite including the builds from https://v2.fangcloud.com/share/7395fd138a1cab496fd4792fe5
with the same results.
Can you help?

xiongyihui, 2018-12-24:
Cloud you upload a recording wav file?

imsunburnt, 2018-12-27:
[hello.zip](https://github.com/respeaker/seeed-voicecard/files/2711284/hello.zip)

imsunburnt, 2018-12-27:
Tried a new stretch-lite build and I get the static problem.  I have uploaded the hello.wav in a zip file.

imsunburnt, 2018-12-27:
I have uploaded the hello.wav file (zipped) in the issues on GitHub and attached it to this email.

xiongyihui, 2018-12-27:
Have you used `alsamixer` or `alsactl` to change the settings the sound card?

[image]

imsunburnt, 2018-12-27:
Yes I have, and I am doing it again right now.  I have examined the wav file using audacity and I can see that mic’s 1,3 and 4 contain the static; mic 2 is better.  I have set gains for mic 1,3,4 to zero and the sound is better but rather faint.  Is there a chat option we can use, if you want?

[image]<https://user-images.githubusercontent.com/948283/50464522-4826ee00-09cc-11e9-9643-06715bf47fe0.png>

xiongyihui, 2018-12-27:
We can use https://gitter.im/seeed-voicecard/Lobby to chat.

Maybe restoring the default settings of the sound card will solve the issue, run
```
cd ~/seeed-voicecard
sudo alsactl restore --file=ac108_asound.state
```

imsunburnt, 2018-12-27:
It looks like the 4mics_hat respeaker board has a problem. I tested a second respeaker board that I have and it is OK.  Here's how I tested (I'll call the board with problem respeaker1 and board with no problem respeaker2):
Swap in respeaker2.  Test over 30 minutes. No problem
Swap in respeaker1.  First test gives good results - no static.  Test 5 times over 3 minutes and the board gets worse, finally producing loud static.
It looks like respeaker1 board may have a problem, possibly heat related (?), since it works ok after being off, but then gets progressively worse over the next few minutes.
I will upload the wav files captured over 3 minutes.

[respeaker-wav-files.zip](https://github.com/respeaker/seeed-voicecard/files/2712399/respeaker-wav-files.zip)

lricher1, 2018-12-31:
i have the same problem since 2 years. I didn't imagine that is board problem !
 Have you find a solution for your respeaker1 ?

xiongyihui, 2019-01-02:
@xyu6

imsunburnt, 2019-01-02:
Hi,

Not sure what your link means?

Pillar1989, 2019-01-14:
I will close this issue, the later you still have questions, you can reopen it

closed 2019-01-14 by Pillar1989