respeaker/seeed-voicecard #124: Left channel 2 mic hat not working

lmore377, 2018-12-26:
I have the ReSpeaker 2 Mic hat and the left microphone dosen't want to work at all. I tried changing some settings within alsamixer but no matter what I change it's still completely silent. I've tried it on a Raspberry Pi 3B+ and Zero W both with fresh installs of Raspbian.

Pillar1989, 2018-12-26:
@xyu6

xyu6, 2018-12-26:
> Mic hat and the left microphone dosen't want to work at all. I tried changing some settings within alsamixer but no matter what I change it's still completely silent. I've tried it on a Raspberry Pi 3B+ and Zero W both with fresh installs of Raspbian.

Hi there, it seems the hardware issue, please take a picture of the mics and drop mail to techsupport@seeed.cc. thanks.

lmore377, 2018-12-26:
I think I'll just return it and order a new one.

closed 2018-12-26 by lmore377

lmore377, 2018-12-26:
I looked at the Amazon listing and the one i bought was actually sold by a company called Keyestudio and not seeed studio.

xiongyihui, 2018-12-26:
Keyestudio copied the respeaker 2-mic hat and still use the name "respeaker 2-mic hat". This is not right. So sad to see that.