respeaker/seeed-voicecard #125: sound always muted

Dloranger, 2019-01-06:
I have a problem that seems to persist through users as pi and root.  The sound levels are always muted after a reboot.

How I get here

1. start with a new sd card image from the latest pi foundation raspbian-lite.
2. manually update and upgrade
3. reboot
4. install the drivers using the commands from the wiki
sudo apt-get update
sudo apt-get upgrade
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh 
sudo reboot

once completed I use alsamixer to set the sound levels to 90% and the use alsactl store to save the settings
 
reboot
alsamixer has the volume at 0% again

Pillar1989, 2019-01-07:
Hi there, it seems the hardware issue, please take a picture of the mics and drop mail to techsupport@seeed.cc. thanks. @xyu6

Pillar1989, 2019-01-14:
I will close this issue, the later you still have questions, you can reopen it

closed 2019-01-14 by Pillar1989