respeaker/seeed-voicecard #127: ReSpeaker_2_mic asoundrc configuration

AndreV2890, 2019-01-10:
Hi

in my project, I use a ReSpeaker_2_mic with the last distribution of Raspian Os. First I try to provide some related information to clarify my system status.

The **_aplay -l_** shows this:
`pi@raspberrypi:~ $ aplay -l 
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

while the _**arecord -l**_:
`pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

and the /etc/asound.conf shows this:
`# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

# use samplerate to resample as speexdsp resample is bad
defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:seeed2micvoicec"
    ipc_key 555555
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:seeed2micvoicec"
        channels 2
    }
    ipc_key 666666
}`

Due to some system constraints I have to switch from stereo to mono. Executing this switching make the system crash because of the impossibility to initialize the system to use a single channel for capture and playback using both the arecord and  aplay terminal command and a custom c code
For instance:
`pi@raspberrypi:~ $ arecord -Dhw:seeed2micvoicec -c 1 -f S16_LE | aplay -Dhw:seeed2micvoicec -c 1 
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
arecord: set_params:1305: Channels count non available
aplay: playback:2787: read error`

while if I try to use stero it works correctly independently from sample rate, format and the other parameters

So my questions are:

1. Is it possible to use the Respeaker 2 mic to acquire a mono signal?
2. Could anyone provide a guide to explain to me how to modify this configuration in the /etc/asound.conf (for instance using the .asoundrc file to override the default behavior)?

jerryyip, 2019-01-10:
Please try `arecord -Dcapture -c 1 -f S16_LE | aplay -Dplayback -c 1 ` or `arecord -Dplughw:seeed2micvoicec -c 1 -f S16_LE | aplay -Dplughw:seeed2micvoicec -c 1`.

AndreV2890, 2019-01-10:
@jerryyip Thanks it seems to work at least in the linux terminal using:
`arecord -Dcapture -c 1 -f S16_LE | aplay -Dplayback -c 1`

Pillar1989, 2019-01-14:
I will close this issue, the later you still have questions, you can reopen it

closed 2019-01-14 by Pillar1989