respeaker/seeed-voicecard #129: Potential problem with install.sh script

treii28, 2019-01-25:
I was having a problem with the script working to install the shield and get it recognizable on my system and discovered one problem that others may run into. There are a couple of lines in the script that check the /boot/config.txt and only modify it if they do not find the modification has been done already. However, they are only checking for the string they, themselves add. The one that was the culprit in my case was:

```
grep -q "dtparam=i2s=on" /boot/config.txt || \
  echo "dtparam=i2s=on" >> /boot/config.txt

```

In my case, this line did not function as designed because my /boot.config already contained a line that read:

#dtparam=i2s=on

so the grep -q returned success and thus the echo did not add the line in an un-commented form.

I thought I would point this out in case anyone else was having problems. Make sure the modifications to /boot/config.txt didn't get skipped due to similar lines existing in the file but being commented out already.

Pernat1y, 2019-01-28:
This one should fix it
https://github.com/respeaker/seeed-voicecard/pull/131

Pillar1989, 2019-02-11:
I will close this issue, the later you still have questions, you can reopen it

closed 2019-02-11 by Pillar1989