respeaker/seeed-voicecard #134: kernel headers报错，Respeaker 2-Mics Pi HAT驱动安装不成功

gaoxueping, 2019-02-02:
Hi 各位好！

我在安装Respeaker 2-Mics Pi HAT驱动时，遇到kernel headers报错，Respeaker 2-Mics Pi HAT驱动安装不成功，有其他人遇到过是怎么回事的么？
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ **_git clone https://github.com/respeaker/seeed-voicecard_**
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 648 (delta 2), reused 0 (delta 0), pack-reused 642
Receiving objects: 100% (648/648), 1.27 MiB | 579.00 KiB/s, done.
Resolving deltas: 100% (401/401), done.
pi@raspberrypi:~ $ **_cd seeed-voicecard_**
pi@raspberrypi:~/seeed-voicecard $ **_sudo ./install.sh_** 
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]          
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease                 
Fetched 25.4 kB in 9s (2,608 B/s)                                                                                                                                    
Reading package lists... Done
Building dependency tree       
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20181112-1).
raspberrypi-kernel-headers is already the newest version (1.20181112-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u4).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.14.79+ -C /lib/modules/4.14.79+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....
cleaning build area...

DKMS: build completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.14.79-v7+ -C /lib/modules/4.14.79-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area...

DKMS: build completed.
Error! echo
Your kernel headers for kernel 4.14.94+ cannot be found at
/lib/modules/4.14.94+/build or /lib/modules/4.14.94+/source.
Error! echo
Your kernel headers for kernel 4.14.94-v7+ cannot be found at
/lib/modules/4.14.94-v7+/build or /lib/modules/4.14.94-v7+/source.
Error! echo
Your kernel headers for kernel 4.9.25-v7+ cannot be found at
/lib/modules/4.9.25-v7+/build or /lib/modules/4.9.25-v7+/source.
Error! echo
Your kernel headers for kernel 4.9.28+ cannot be found at
/lib/modules/4.9.28+/build or /lib/modules/4.9.28+/source.
Error! echo
Your kernel headers for kernel 4.9.28-v7+ cannot be found at
/lib/modules/4.9.28-v7+/build or /lib/modules/4.9.28-v7+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) eed3855] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ **_sudo reboot_**
Connection to 192.168.186.158 closed by remote host.
Connection to 192.168.186.158 closed.
Pings-MacBook-Air:~ xueping.gao$ ssh -l pi 192.168.186.158
pi@192.168.186.158's password: 
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Feb  2 09:24:49 2019
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ **_cd seeed-voicecard/_**
pi@raspberrypi:~/seeed-voicecard $ **_aplay -l_**
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ **_arecord -l_**
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ **_uname -a_**
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ **_lsb_release -a_**
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch
pi@raspberrypi:~/seeed-voicecard $ 
pi@raspberrypi:~/seeed-voicecard $ **_cat /proc/version_**
Linux version 4.14.79-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1159 SMP Sun Nov 4 17:50:20 GMT 2018
pi@raspberrypi:~/seeed-voicecard $ 

谁能帮我看看是哪里出了问题呢？先谢过了！

freakadings, 2019-02-10:
got the same problem :/

```
Error! echo
Your kernel headers for kernel 4.14.97+ cannot be found at
/lib/modules/4.14.97+/build or /lib/modules/4.14.97+/source.
Error! echo
Your kernel headers for kernel 4.14.97-v7+ cannot be found at
/lib/modules/4.14.97-v7+/build or /lib/modules/4.14.97-v7+/source.
mkdir: das Verzeichnis „/etc/voicecard“ kann nicht angelegt werden: Die Datei existiert bereits
git init
Bestehendes Git-Repository in /etc/voicecard/.git/ neuinitialisiert
git add --all
git commit -m "origin configures"
Auf Branch master
nichts zu committen, Arbeitsverzeichnis unverändert
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

```

Pillar1989, 2019-02-11:
#100 
Please refer this issue.

Pillar1989, 2019-02-11:
Duplicate of #100

closed 2019-03-25 by Pillar1989