respeaker/seeed-voicecard #136: pcm.multipleapps not working

JonahKr, 2019-02-16:
Hey there 👋 ,

Atm I am trying to setup the Respeaker 4-Mic Array on a Raspberry Pi 3B+ and it works so far.
But to make it usable for multiple services/actions/programms i have to change the `asound.conf` as far i am concerned. 

Therefore i changed it to enable `dsnoop` :

``` # The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

# use samplerate to resample as speexdsp resample is bad
defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "ac108"
}

pcm.playback {
    type plug
    slave.pcm "hw:ALSA"
}

# pcm.dmixed {
#     type dmix
#     slave.pcm "hw:0,0"
#     ipc_key 555555
# }

pcm.ac108 {
    type plug
    slave.pcm "multiapps"
}

pcm.multiapps {
    type dsnoop
    slave.pcm "hw:1,0"
    ipc_key 666666
}
``` 
But this doesn't work and I still get the error : _ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave_

I hope that one of you can help me with this problem.
Thank you in advance,

Jonah

xiongyihui, 2019-02-19:
May the IPC key `666666` is being used by other applications. Try another number (for example, 666667) to see if it is the problem.

JonahKr, 2019-02-21:
I tried this @xiongyihui  and it didn't work 🤔 
still: 
`ALSA lib pcm_direct.c:1765:(snd1_pcm_direct_parse_open_conf) Unknown field ac108-slavepcm
arecord: main:788: audio open error: Invalid argument`

JonahKr, 2019-02-23:
i solved it:
`sudo alsactl --file=/etc/voicecard/ac108_asound.state store` didn't work because of whatever.
after i run the command and restarted everything, now it works fine.

closed 2019-02-23 by JonahKr