respeaker/seeed-voicecard #137: install on kali linux

cogmeta, 2019-03-03:
Can i install this driver on kali linux?

DulyHub, 2019-03-04:
Yes you can!

git clone https://github.com/respeaker/seeed-voicecard
cd seeed-voicecard
sudo ./install.sh 
sudo reboot

cogmeta, 2019-03-05:
@DulyHub. Just a humble request. Before randomly commenting on issues, please make sure you understand what the issue is talking about. What you suggested (or rather copied directly) works for raspberian not for anything else.  Currently seeed voicecard has no support for anything other than raspberian. 

so, @DulyHub. Read and try youself before you write random things.

Pillar1989, 2019-03-05:
We also support Arch linux

cogmeta, 2019-03-05:
@Pillar1989 I was hoping a reply from you :) but, it installation does not work as of now.

Your kernel headers for kernel 4.14.93-Re4son-v8+ cannot be found

baorepo, 2019-03-05:
Linux-4.14.93-re4son-v8 + is maintained by Keil linux. You might ask their community, In addition, I would like to ask, what is the reason for you to give up being raspberian.

cogmeta, 2019-03-05:
We deploy lot of these devices with complete disk encryption and remote boot using mandos which works perfectly on linux. We have not had luck running mados yet on raspberian.

Pillar1989, 2019-03-06:
@cogmeta  Raspberry Pi foundation does not support this Keil linux. I am sorry that we cannot maintain it unless Keil linux actively supports it

closed 2019-03-25 by Pillar1989