respeaker/seeed-voicecard #139: ReSpeaker 2-Mics with stereo speakers

dubaleeiro, 2019-03-05:
Hi, what is the best war to connect 2x 5W 8 Ohm speakers to the board? Audio JAC? GPIO pins? JST audio connector is just for mono speakers...
Thank you!

Pillar1989, 2019-03-06:
AUX interface support stereo speakers, audio connector is just for mono speakers

dubaleeiro, 2019-03-06:
Hi, which one you mean AUX interface? The JST speaker out or 3.5mm Audio JAC? Thank you!

dubaleeiro, 2019-03-15:
Dear @Pillar1989, could you please  explain again how I can connect stereo speakers? Thank you

Pillar1989, 2019-03-18:
3.5mm Audio JAC

ebs54, 2019-03-24:
Hi, I like to use the ReSpeaker 2-Mics in a Raspberry Pi zero W with Volumio OS. The Assistent can select varios types of HAT but not the ReSpeaker 2-Mics. 
Can you help me?
Thanks
Ebs

dubaleeiro, 2019-03-24:
Same here! I have sent an email to them but so far I got no answer...

Pillar1989, 2019-03-25:
What is Volumio OS? I did not see this release in http://www.raspberrypi.org

ebs54, 2019-03-25:
See volumio.org

Pillar1989, 2019-03-25:
So wouldn't it be better to discuss this with https://volumio.org/forum/ ?

ebs54, 2019-03-25:
There I could not find your pHAT.If your pHAT is compatible with an other well known, this is very helpfull.      Von meinem Samsung Galaxy Smartphone gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/respeaker/seeed-voicecard/issues/139#issuecomment-476070977",
"url": "https://github.com/respeaker/seeed-voicecard/issues/139#issuecomment-476070977",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2019-04-16 by Pillar1989