respeaker/seeed-voicecard #140: aarch64 not supported?

dony71, 2019-03-07:
I compile on ubuntu 18.04.2 aarch64 (kernel Linux ubuntu 4.15.0-1032-raspi2)
After some modification on install.sh , I manage to compile successfully and installed
However while loading module in dmesg shows

> snd_soc_seeed_voicecard:`
 module verification failed: signature and/or required key missing - tainting kernel

how to make this aarch64 compatible?

Pillar1989, 2019-03-11:
I haven't tested on aarch64. Where can I find an image of raspberry pi aarch64? @dony71

hbehrendt, 2019-03-11:
I have the same issue under debian9 arm64 from https://people.debian.org/~gwolf/raspberrypi3/20190206/20190206-raspberry-pi-3-buster-PREVIEW.img.xz and https://wiki.debian.org/RaspberryPi3
All on a raspi 3 b+
hope it helps 
[make.log](https://github.com/respeaker/seeed-voicecard/files/2951752/make.log)

Pillar1989, 2019-03-11:
Sorry, this is not the official  image of raspberry pi. There are too many personal  images, and there are many problems with the  image itself. Please forgive us if we cannot support all of them.

hbehrendt, 2019-03-11:
Hi, as i understood it is a Official upcoming Debian Buster release of
the arm64 branch. I had the same issue with ubuntu arm64. As both use a
official kernel and dbkg to build it, it might be a good chance to have
even a look at it. The debian Kernel looks like the official 4.4.19 arm
kernel. The issues i had where with the Kernel drivers, not with any
flavour id userland programms.

Kind regars Helfried Behrendt

dony71, 2019-03-11:
Ubuntu aarch64 can be downloaded from here :
http://cdimage.ubuntu.com/releases/18.04/release/ubuntu-18.04.2-preinstalled-server-arm64+raspi3.img.xz

turmary, 2019-03-14:
Hi @hbehrendt 
We tried to run seeed-voicecard on debian9 arm64 platform, but it failed cause by missing of dtoverlay support in bootloader or system.

Hi @dony71 
Could you get dtoverlay in Ubuntu aarch64 ?
If it's not, Ubuntu aarch64 also can't be supported.

bkchr, 2019-03-16:
Hey, for me it is working with aarch64.

norbusan, 2019-03-23:
The problem with the image from Debian buster is that `CONFIG_SND_SIMPLE_CARD` is not enabled in the config. This prevents loading of the modules due to missing symbols.

Then there is the problem with dtoverlay

closed 2019-03-25 by Pillar1989

norbusan, 2019-03-25:
Just to mention that when booting with Ubuntu's raspi3 image which has the above option set, I could get the hat to be recognized.