respeaker/seeed-voicecard #142: Raspberry pi 3 b+ not detecting Respeaker 4-mic array

amitkray31, 2019-03-12:
I have **4.19.27-v7+ Kernel** installed but it is using the Kernel version 4.14.98-v7+ during installation of Respeaker 4-mic array driver. I think it is creating the problem. I have tried to install Respeaker 4-mic array driver, removing the all old Kernels `sudo rm -rf 4.14.98+` ,  `sudo rm -rf 4.14.98-v7+`
`sudo rm -rf 4.19.27+`. But as I run the command  `sudo ./install.sh 4mic` then first it reinstalling the old Kernels and after that using those old kernels to build it.
**Can anyone help me to solve this problem?**
`pi@raspberrypi:~ $ git clone https://github.com/respeaker/seeed-voicecard.git`
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 648 (delta 2), reused 6 (delta 2), pack-reused 642
Receiving objects: 100% (648/648), 1.27 MiB | 557.00 KiB/s, done.
Resolving deltas: 100% (401/401), done.
`pi@raspberrypi:~ $ cd seeed-voicecard`
`pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh 4mic`
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done                      
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20190215-1).
The following NEW packages will be installed:
  raspberrypi-kernel-headers
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 16.1 MB of archives.
After this operation, 105 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-kernel-headers armhf 1.20190215-1 [16.1 MB]
Fetched 16.1 MB in 14s (1,109 kB/s)                                            
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 35579 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20190215-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20190215-1) ...
Setting up raspberrypi-kernel-headers (1.20190215-1) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.11.0-3+deb9u4).
The following additional packages will be installed:
  fontconfig-config fonts-dejavu-core libasyncns0 libavcodec57 libavresample3
  libavutil55 libcairo2 libdrm-amdgpu1 libdrm-freedreno1 libdrm-nouveau2
  libdrm-radeon1 libflac8 libfontconfig1 libgl1-mesa-dri libgl1-mesa-glx
  libglapi-mesa libgsm1 libice6 libjack-jackd2-0 libllvm3.9 libmp3lame0
  libogg0 libopenjp2-7 libopus0 libpixman-1-0 libpulse0 libsensors4 libshine3
  libsm6 libsnappy1v5 libsndfile1 libsoxr0 libspeex1 libspeexdsp1
  libswresample2 libtheora0 libtwolame0 libtxc-dxtn-s2tc libva-drm1
  libva-x11-1 libva1 libvdpau-va-gl1 libvdpau1 libvorbis0a libvorbisenc2
  libvpx4 libwavpack1 libwebp6 libwebpmux2 libx11-xcb1 libx264-148 libx265-95
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0
  libxcb-shm0 libxcb-sync1 libxdamage1 libxfixes3 libxi6 libxrender1
  libxshmfence1 libxtst6 libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0
  mesa-va-drivers mesa-vdpau-drivers read-edid va-driver-all vdpau-driver-all
  x11-common
Suggested packages:
  python3-apport menu libi2c-dev python-smbus jackd2 opus-tools pulseaudio
  lm-sensors speex
The following NEW packages will be installed:
  dkms fontconfig-config fonts-dejavu-core i2c-tools libasound2-plugins
  libasyncns0 libavcodec57 libavresample3 libavutil55 libcairo2 libdrm-amdgpu1
  libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libflac8 libfontconfig1
  libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgsm1 libice6
  libjack-jackd2-0 libllvm3.9 libmp3lame0 libogg0 libopenjp2-7 libopus0
  libpixman-1-0 libpulse0 libsensors4 libshine3 libsm6 libsnappy1v5
  libsndfile1 libsoxr0 libspeex1 libspeexdsp1 libswresample2 libtheora0
  libtwolame0 libtxc-dxtn-s2tc libva-drm1 libva-x11-1 libva1 libvdpau-va-gl1
  libvdpau1 libvorbis0a libvorbisenc2 libvpx4 libwavpack1 libwebp6 libwebpmux2
  libx11-xcb1 libx264-148 libx265-95 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0
  libxcb-present0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxdamage1
  libxfixes3 libxi6 libxrender1 libxshmfence1 libxtst6 libxvidcore4
[... 441 lines cut ...]
[master (root-commit) 01b5701] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
`pi@raspberrypi:~/seeed-voicecard $ sudo reboot`

`pi@raspberrypi:~ $ arecord -l`
**** List of CAPTURE Hardware Devices ****
`pi@raspberrypi:~ $ arecord -L`
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
ac108
pi@raspberrypi:~ $

Pillar1989, 2019-03-12:
Duplicate of #140

amitkray31, 2019-03-12:
my problem have been solved

closed 2019-03-12 by amitkray31

kajo222, 2019-08-18:
How you solved problem? - I have exacly the same problem.

amitkray31, 2019-08-18:
*That problem occurred due to the mismatch of the kernel version of
Raspbian OS.* *Do not run the command **sudo apt-get install rpi-update*.
Raspbian kernel automatically upgrades through the *sudo apt-get update & &
sudo apt-get upgrade*. *Choose supported version of kernel. *

kajo222, 2019-08-19:
Which one is compatible because I tried 4, even 2 years earlier and still have a problem?

amitkray31, 2019-08-19:
Run the command `uname -r` and show me the output.

kajo222, 2019-08-19:
I have been in work, sorry - 4.19.58-v7+

amitkray31, 2019-08-19:
Rollback to the version 4.19.57-v7+

amitkray31, 2019-08-19:
Sorry, ignore the previous instruction. Just execute `sudo apt-get update && sudo apt-get upgrade` on terminal. After this **reboot** your pi. After Reboot execute  `uname -r`  on terminal then you will see this **4.19.66-v7+** . Now execute `arecord -l` on terminal  and show me the output

kajo222, 2019-08-19:
Finally Updated 
[image]

amitkray31, 2019-08-19:
Do you reinstall the respeaker?

amitkray31, 2019-08-19:
Reinstall the respeaker software

kajo222, 2019-08-19:
still the same

kajo222, 2019-08-19:
Is it problem to you to send me image from win32diskimager with installed
driver?

wt., 20 sie 2019, 00:07 użytkownik Kacper korczak <
kacper.korczak123@gmail.com> napisał:

kajo222, 2019-08-19:
I am using Raspberry pi3

wt., 20 sie 2019, 00:08 użytkownik Kacper korczak <
kacper.korczak123@gmail.com> napisał:

kajo222, 2019-08-19:
I don't think it sounded nice in your language, but I didn't have that
purpose

wt., 20 sie 2019, 00:08 użytkownik Kacper korczak <
kacper.korczak123@gmail.com> napisał:

amitkray31, 2019-08-19:
No, it will not be a problem.
But you can also do it on your current Raspbian image.

kajo222, 2019-08-20:
We can try install it on my Raspbian and thanks for help :)

amitkray31, 2019-08-20:
Do you reinstall the respeaker software?

kajo222, 2019-08-20:
Yes

kajo222, 2019-08-20:
This same output

wt., 20 sie 2019, 07:28 użytkownik Kacper korczak <
kacper.korczak123@gmail.com> napisał:

amitkray31, 2019-08-20:
Do you connect your respeaker module on top of your pi?

kajo222, 2019-08-20:
Yes, and i check it and i see it in i2c and RGB leds working

amitkray31, 2019-08-20:
Reinstall  the respeaker software again and **send me the terminal output to me**

kajo222, 2019-08-20:
Te same output

kajo222, 2019-08-20:
Ok, you want to se error od something

wt., 20 sie 2019, 07:38 użytkownik Kacper korczak <
kacper.korczak123@gmail.com> napisał:

amitkray31, 2019-08-20:
I need that ouput message

kajo222, 2019-08-20:
Yeah i taked Raspberry to Work and wait while i must configure

kajo222, 2019-08-20:
kacper@kacper-INVALID:~$ ssh pi@10.10.0.59
The authenticity of host '10.10.0.59 (10.10.0.59)' can't be established.
ECDSA key fingerprint is SHA256:BBUKLmNZlzeSFxQdabqSuWoy79Fx+6U8114vep2Ldiw.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '10.10.0.59' (ECDSA) to the list of known hosts.
pi@10.10.0.59's password:
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Aug 20 00:15:57 2019 from 192.168.1.97

SSH is enabled and the default password for the 'pi' user has not been
changed.
This is a security risk - please login as the 'pi' user and type 'passwd'
to set a new password.

pi@raspberrypi:~ $ cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Pobieranie:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
[15,0 kB]
Stary:2 http://archive.raspberrypi.org/debian stretch InRelease
Pobrano 15,0 kB w 1s (12,6 kB/s)
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
All packages are up to date.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
raspberrypi-kernel is already the newest version (1.20190819~stretch-1).
raspberrypi-kernel-headers is already the newest version
(1.20190819~stretch-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej
wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u4).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej
wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.66-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.19.66-v7+ -C /lib/modules/4.19.66-v7+/build
M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area......

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.
mkdir: nie można utworzyć katalogu „/etc/voicecard”: Plik istnieje
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
pi@raspberrypi:~/seeed-voicecard $

wt., 20 sie 2019 o 07:56 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

amitkray31, 2019-08-20:
Ok. After rebooting your pi run **` arecord -L && aplay -L`** and send me the output

kajo222, 2019-08-20:
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
ac108
dmixer
ac101
pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
ac108
dmixer
ac101
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dmix:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Hardware device with all software conversions

amitkray31, 2019-08-20:
Are you using Respeaker 4-mic aray?

amitkray31, 2019-08-20:
Execute `arecord -l`  on terminal and send me the output

amitkray31, 2019-08-20:
_You have some different problem here and it is not due to the kernel_
_see the bolded lines :-_ 

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   **- No original module exists within this kernel**
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   **- No original module exists within this kernel**
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   **- No original module exists within this kernel**
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/

kajo222, 2019-08-20:
6-mic

kajo222, 2019-08-20:
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~ $ ^C
pi@raspberrypi:~ $

wt., 20 sie 2019 o 08:41 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

amitkray31, 2019-08-20:
See here https://github.com/respeaker/seeed-voicecard/issues/149

kajo222, 2019-08-20:
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Stary:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Stary:2 http://archive.raspberrypi.org/debian stretch InRelease
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
All packages are up to date.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
raspberrypi-kernel is already the newest version (1.20190819~stretch-1).
raspberrypi-kernel-headers is already the newest version
(1.20190819~stretch-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej
wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u4).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej
wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.66-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.19.66+ -C /lib/modules/4.19.66+/build
M=/var/lib/dkms/seeed-voicecard/0.3/build.......
cleaning build area......

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/bcm/

depmod....

DKMS: install completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.19.66-v7+ -C /lib/modules/4.19.66-v7+/build
M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area....

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.
Error! echo
Your kernel headers for kernel 4.19.66-v7l+ cannot be found at
/lib/modules/4.19.66-v7l+/build or /lib/modules/4.19.66-v7l+/source.
mkdir: nie można utworzyć katalogu „/etc/voicecard”: Plik istnieje
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

wt., 20 sie 2019 o 08:41 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

kajo222, 2019-08-20:
I remembered that I tried to fix it and modified it in the install
kernels=${uname_r}

kajo222, 2019-08-20:
install.sh and after that error dissaperaed, but still dont work

wt., 20 sie 2019 o 08:49 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

kajo222, 2019-08-20:
i should do sudo apt-get -y --reinstall install raspberrypi-kernel?

wt., 20 sie 2019 o 08:50 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

amitkray31, 2019-08-20:
**Execute this on terminal** 
`cd seeed-voicecard && sudo ./uninstall.sh && cd .. && sudo rm -rf seeed-voicecard && sudo reboot`
**After rebooting** 
`git clone https://github.com/respeaker/seeed-voicecard.git && cd seeed-voicecard && sudo ./install.sh 6mic && sudo reboot`
After this run **arecord -l** and see what  happens.
This is all I can say.

kajo222, 2019-08-20:
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh 6mic
Stary:1 http://archive.raspberrypi.org/debian stretch InRelease
Stary:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
All packages are up to date.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
raspberrypi-kernel is already the newest version (1.20190819~stretch-1).
raspberrypi-kernel-headers is already the newest version
(1.20190819~stretch-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej
wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności
Odczyt informacji o stanie... Gotowe
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u4).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej
wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.19.66+ -C /lib/modules/4.19.66+/build
M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area......

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
   - Storing in
/var/lib/dkms/seeed-voicecard/original_module/4.19.66+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66+/kernel/sound/soc/codecs/snd-soc-ac108.ko
   - Storing in
/var/lib/dkms/seeed-voicecard/original_module/4.19.66+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - Found
/lib/modules/4.19.66+/kernel/sound/soc/bcm/snd-soc-seeed-voicecard.ko
   - Storing in
/var/lib/dkms/seeed-voicecard/original_module/4.19.66+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/bcm/

depmod....

DKMS: install completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.19.66-v7+ -C /lib/modules/4.19.66-v7+/build
M=/var/lib/dkms/seeed-voicecard/0.3/build.......
cleaning build area......

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/

depmod......

DKMS: install completed.
Error! echo
Your kernel headers for kernel 4.19.66-v7l+ cannot be found at
/lib/modules/4.19.66-v7l+/build or /lib/modules/4.19.66-v7l+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) bec4fc4] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink
/etc/systemd/system/sysinit.target.wants/seeed-voicecard.service →
/lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
pi@raspberrypi:~/seeed-voicecard $

kajo222, 2019-08-20:
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~ $

wt., 20 sie 2019 o 09:27 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

amitkray31, 2019-08-20:
**The only way that is left** is to download latest Raspbian image, load it Raspberry pi, configure it. Run `sudo apt-get update && sudo apt-get upgrade` **( do not run rpi-update)**  `sudo apt-get install git && git clone https://github.com/respeaker/seeed-voicecard.git && cd seeed-voicecard && sudo ./install.sh 6mic && sudo reboot`

**I can not assure you that it will work.**  But you can try.
Good luck

kajo222, 2019-08-20:
i tried

kajo222, 2019-08-20:
Can you send me image?

wt., 20 sie 2019 o 09:52 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

amitkray31, 2019-08-20:
I will try my best to send you that disk image. It will take some time.

kajo222, 2019-08-20:
Ok, very thanks to you

kajo222, 2019-08-20:
maybe its help

pi@raspberrypi:~ $ cat /var/log/seeed-voicecard.log
+ dtparam i2c_arm=on
+ modprobe i2c-dev
+ dtparam spi=on
No protocol specified
No protocol specified
+ _VER_RUN=
+ CONFIG=/boot/config.txt
++ i2cdetect -y 1 0x1a 0x1a
++ egrep '(1a|UU)'
++ awk '{print $2}'
+ is_1a=UU
++ egrep '(35|UU)'
++ i2cdetect -y 1 0x35 0x35
++ awk '{print $2}'
+ is_35=UU
++ i2cdetect -y 1 0x3b 0x3b
++ egrep '(3b|UU)'
++ awk '{print $2}'
+ is_3b=UU
+ RPI_HATS='seeed-2mic-voicecard seeed-4mic-voicecard seeed-8mic-voicecard'
+ overlay=
+ '[' xUU '!=' x ']'
+ '[' xUU == x ']'
+ '[' xUU '!=' x ']'
+ '[' xUU == x ']'
+ '[' xUU '!=' x ']'
+ '[' xUU '!=' x ']'
+ echo 'install 6mic'
install 6mic
+ overlay=seeed-8mic-voicecard
+ asound_conf=/etc/voicecard/asound_6mic.conf
+ asound_state=/etc/voicecard/ac108_6mic.state
+ '[' seeed-8mic-voicecard ']'
+ echo Install seeed-8mic-voicecard ...
Install seeed-8mic-voicecard ...
+ rm /etc/asound.conf
+ rm /var/lib/alsa/asound.state
++ get_kernel_version
++ local ZIMAGE IMG_OFFSET
++ _VER_RUN=
++ '[' -z '' ']'
++ ZIMAGE=/boot/kernel.img
+++ head -n 1
+++ cut -d : -f 1
+++ LC_ALL=C
+++ grep -abo $'\037\213\b' /boot/kernel.img
++ IMG_OFFSET=18092
+++ dd if=/boot/kernel.img obs=64K ibs=4 skip=4523
+++ strings
+++ zcat
+++ awk '{ print $3; }'
+++ grep -a -m1 'Linux version'

gzip: stdout: Broken pipe
dd: zapis do 'standardowe wyjście': Przerwany potok
933888+0 przeczytanych rekordów
56+0 zapisanych rekordów
3670016 bajtów (3,7 MB, 3,5 MiB), 4,77969 s, 768 kB/s
++ _VER_RUN=4.19.66+
++ echo 4.19.66+
++ return 0
+ kernel_ver=4.19.66+
+ [[ 4.19.66+ =~ ^4\.19.*$ ]]
+ for i in $RPI_HATS
+ '[' seeed-2mic-voicecard == seeed-8mic-voicecard ']'
+ echo Uninstall seeed-2mic-voicecard ...
Uninstall seeed-2mic-voicecard ...
+ do_overlay seeed-2mic-voicecard 1
+ ov=seeed-2mic-voicecard
+ RET=1
+ DEFAULT=--defaultno
+ CURRENT=0
++ get_overlay seeed-2mic-voicecard
++ ov=seeed-2mic-voicecard
++ grep -q -E '^dtoverlay=seeed-2mic-voicecard' /boot/config.txt
++ echo 1
+ '[' 1 -eq 0 ']'
+ '[' 1 -eq 0 ']'
+ '[' 1 -eq 0 ']'
+ '[' 1 -eq 1 ']'
+ sed /boot/config.txt -i -e
's/^dtoverlay=seeed-2mic-voicecard/#dtoverlay=seeed-2mic-voicecard/'
+ STATUS=disabled
+ for i in $RPI_HATS
+ '[' seeed-4mic-voicecard == seeed-8mic-voicecard ']'
+ echo Uninstall seeed-4mic-voicecard ...
Uninstall seeed-4mic-voicecard ...
+ do_overlay seeed-4mic-voicecard 1
+ ov=seeed-4mic-voicecard
+ RET=1
+ DEFAULT=--defaultno
+ CURRENT=0
++ get_overlay seeed-4mic-voicecard
++ ov=seeed-4mic-voicecard
++ grep -q -E '^dtoverlay=seeed-4mic-voicecard' /boot/config.txt
++ echo 1
+ '[' 1 -eq 0 ']'
+ '[' 1 -eq 0 ']'
+ '[' 1 -eq 0 ']'
+ '[' 1 -eq 1 ']'
+ sed /boot/config.txt -i -e
's/^dtoverlay=seeed-4mic-voicecard/#dtoverlay=seeed-4mic-voicecard/'
+ STATUS=disabled
+ for i in $RPI_HATS
+ '[' seeed-8mic-voicecard == seeed-8mic-voicecard ']'
+ do_overlay seeed-8mic-voicecard 0
+ ov=seeed-8mic-voicecard
+ RET=0
+ DEFAULT=--defaultno
+ CURRENT=0
++ get_overlay seeed-8mic-voicecard
++ ov=seeed-8mic-voicecard
++ grep -q -E '^dtoverlay=seeed-8mic-voicecard' /boot/config.txt
++ echo 0
+ '[' 0 -eq 0 ']'
+ DEFAULT=
+ CURRENT=1
+ '[' 0 -eq 1 ']'
+ '[' 0 -eq 0 ']'
+ sed /boot/config.txt -i -e
's/^#dtoverlay=seeed-8mic-voicecard/dtoverlay=seeed-8mic-voicecard/'
+ grep -q -E '^dtoverlay=seeed-8mic-voicecard' /boot/config.txt
+ STATUS=enabled
+ dtoverlay seeed-8mic-voicecard
No protocol specified
* Failed to apply overlay '2_seeed-8mic-voicecard' (kernel)
No protocol specified
+ true
+ echo 'create seeed-8mic-voicecard asound configure file'
create seeed-8mic-voicecard asound configure file
+ ln -s /etc/voicecard/asound_6mic.conf /etc/asound.conf
+ echo 'create seeed-8mic-voicecard asound status file'
create seeed-8mic-voicecard asound status file
+ ln -s /etc/voicecard/ac108_6mic.state /var/lib/alsa/asound.state
+ alsactl restore
Found hardware: "bcm2835_alsa" "Broadcom Mixer" "" "" ""
Hardware is initialized using a generic method
alsactl: set_control:1325: failed to obtain info for control #6 (No such
file or directory)
+ amixer cset numid=3 1
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=3,step=0
  : values=1

wt., 20 sie 2019 o 10:03 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

kajo222, 2019-08-20:
AND i2c detect

pi@raspberrypi:~ $  sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- UU -- -- -- -- -- UU -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

wt., 20 sie 2019 o 11:46 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):