respeaker/seeed-voicecard #146: AVS+KITT.AI compiled-ok but test-failed with RPiZero/3B+ReSpeaker 2-Mics Pi HAT

iwwork, 2019-03-21:
AVS+KITT.AI compiled-ok but test-failed with RPiZero/3B+ReSpeaker 2-Mics Pi HAT · Issue #1286 · alexa/avs-device-sdk
https://github.com/alexa/avs-device-sdk/issues/1286

Following this link:
Compilation hangs on bleutooth · Issue #1014 · alexa/avs-device-sdk
#1014 (comment)
@henriz has managed to run RPiZero/3B+ReSpeaker 2-Mics Pi HAT with avs and it works fine.

Of cuz, Kitt.AI should replace Sensory.

Then found the link:
Poor documentation about using KITT.AI with AVS SDK 1.11 · Issue #1169 · alexa/avs-device-sdk
#1169 (comment)
@jmfrancofraiz work with kitt.ai in my AIY Voice Kit v1.
downloaded the script:
https://github.com/alexa/avs-device-sdk/files/2761089/scripts.zip

So
1 choose Raspbian Stretch Lite Ver:20181113 (this is good with RPi3B for alexa) from
https://www.raspberrypi.org/downloads/raspbian/

2 install driver for ReSpeaker 2-Mics Pi HAT on RPiZero/3B
https://github.com/respeaker/seeed-voicecard.git

3 upgrade all

4 run the scripts
Poor documentation about using KITT.AI with AVS SDK 1.11 · Issue #1169 · alexa/avs-device-sdk
#1169 (comment)
downloaded the script:
https://github.com/alexa/avs-device-sdk/files/2761089/scripts.zip

5 the result is compiled-ok but test-failed

What behavior are you observing?

The following tests FAILED:
282 - TaskThreadTest.testMoniker (Failed)
368 - DirectiveProcessorTest.testSetDialogRequestIdCancelsOutstandingDirectives (Failed)
371 - DirectiveProcessorTest.testAudioAndVisualIsBlockingAudio (Failed)
373 - DirectiveProcessorTest.testReleaseOneMedium (Failed)
374 - DirectiveProcessorTest.TestBlockingQueuedDirectivIsBlocking (Failed)
375 - DirectiveProcessorTest.TestNonBlockingQueuedDirectivIsNotBlocking (Failed)
391 - DirectiveSequencerTest.testBlockingThenNonDialogDirective (Failed)
394 - DirectiveSequencerTest.testThatBargeInDropsSubsequentDirectives (Failed)
397 - DirectiveSequencerTest.testAddDirectiveHandlersWhileHandlingDirectives (Failed)
398 - DirectiveSequencerTest.testHandleBlockingThenImmediatelyThenNonBockingOnSameDialogId (Failed)
608 - KittAiKeyWordTest.invalidStream (Failed)
609 - KittAiKeyWordTest.incompatibleEndianness (Failed)
610 - KittAiKeyWordTest.getExpectedNumberOfDetectionsInFourAlexasAudioFileForOneObserver (Failed)
611 - KittAiKeyWordTest.getExpectedNumberOfDetectionsInFourAlexasAudioFileForTwoObservers (Failed)
612 - KittAiKeyWordTest.getExpectedNumberOfDetectionsInAlexaStopAlexaJokeAudioFileForOneObserver (Failed)
613 - KittAiKeyWordTest.getExpectedNumberOfDetectionsInAlexaStopAlexaJokeAudioFileForTwoObservers (Failed)
614 - KittAiKeyWordTest.getActiveState (Failed)
615 - KittAiKeyWordTest.getStreamClosedState (Failed)
844 - AudioPlayerTest.testPlaybackStartedCallbackAfterFocusLost (Failed)
1087 - MediaPlayerTest.testStartPlayWaitForEnd (Failed)
1088 - MediaPlayerTest.testStartPlayForUrl (Failed)
1089 - MediaPlayerTest.testConsecutiveSetSource (Failed)
1090 - MediaPlayerTest.testStartPlayWaitForEndStartPlayAgain (Failed)
1091 - MediaPlayerTest.testStopPlay (Failed)
1092 - MediaPlayerTest.testStartPlayCallAfterStopPlay (Failed)
1093 - MediaPlayerTest.testStartPlayCallAfterStopPlayDifferentSource (Failed)
1094 - MediaPlayerTest.testPauseDuringPlay (Failed)
1095 - MediaPlayerTest.testResumeAfterPauseThenStop (Failed)
1096 - MediaPlayerTest.testStopAfterPause (Failed)
1097 - MediaPlayerTest.testPauseAfterPause (Failed)
1098 - MediaPlayerTest.testResumeAfterPlay (Failed)
1099 - MediaPlayerTest.testGetOffsetInMilliseconds (Failed)
1101 - MediaPlayerTest.testGetOffsetWhenStoppedFails (Failed)
1102 - MediaPlayerTest.testGetOffsetWhenPaused (Failed)
1103 - MediaPlayerTest.testPlayingTwoAttachments (Failed)
1104 - MediaPlayerTest.testUnsteadyReads (Failed)
1105 - MediaPlayerTest.testRecoveryFromPausedReads (Failed)
1106 - MediaPlayerTest.testStartPlayWithUrlPlaylistWaitForEnd (Failed)
1107 - MediaPlayerTest.testSetOffsetSeekableSource (Failed)
1110 - MediaPlayerTest.testRepeatAttachment (Failed)
1111 - MediaPlayerTest.testSetVolumePlays (Failed)
1112 - MediaPlayerTest.testAdjustVolumePlaysDuringPlay (Failed)
1113 - MediaPlayerTest.testAdjustVolumePlays (Failed)
1114 - MediaPlayerTest.testAdjustVolumeOutOfBounds (Failed)
1115 - MediaPlayerTest.testSetMutePlays (Failed)
1118 - MediaPlayerTest.testReadTags (Failed)
1119 - MediaPlayerTest.testConsecutiveSameApiCalls (Failed)
1120 - MediaPlayerTest.testImmediatePause (Failed)
1121 - MediaPlayerTest.multiplePlayAndSetSource (Failed)
1123 - MediaPlayerTest.doublePause (Failed)
1124 - MediaPlayerTest.resumeWhenPlaying (Failed)
1125 - MediaPlayerTest.resumeWhenStopped (Failed)
1126 - MediaPlayerTest.newSetSourceLeadsToStoppedCallback (Failed)
1127 - MediaPlayerTest.resumeAfterPauseWithPendingPlay (Failed)
1128 - MediaPlayerTest.testRepeatPlayForUrl (Failed)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
Provide the steps to reproduce the issue, if applicable:

1 choose Raspbian Stretch Lite Ver:20181113 (this is good with RPi3B for alexa) from
https://www.raspberrypi.org/downloads/raspbian/

2 install driver for ReSpeaker 2-Mics Pi HAT on RPiZero
https://github.com/respeaker/seeed-voicecard.git

3 upgrade all

4 run the scripts
Poor documentation about using KITT.AI with AVS SDK 1.11 · Issue #1169 · alexa/avs-device-sdk
#1169 (comment)
downloaded the script:
https://github.com/alexa/avs-device-sdk/files/2761089/scripts.zip

5 the result is compiled-ok but test-failed

Tell us about your environment:

What version of the AVS Device SDK are you using?
the newest one

Tell us what hardware you're using:
[x ] Raspberry Pi Zero/3B

Tell us about your OS (Type & version):
[x ] Raspbian Stretch

Pillar1989, 2019-03-25:
Why copy those issue? @iwwork

iwwork, 2019-03-25:
seeking solution @Pillar1989

Pillar1989, 2019-03-25:
@iwwork But I don't think it's a good place to talk about it. I didn't see a seeed-voicecard description of the issue with the software and hardware

closed 2019-04-16 by Pillar1989