respeaker/seeed-voicecard #148: Errors with Alsa: Cannot open shared library + Cannot find rate converter

schneekluth, 2019-04-03:
I don't know if this belongs here or in the alsa project. I've installed the Respeaker 2mic phat and it's working perfectly. I'm trying to install this project called [TalkiePi](https://github.com/CustomMachines/talkiepi) and the speaker is working fine but microphone of the Respeaker seems to have an issue. In the project's code the button of the phat is used to record audio but whenever I press it the following error appears:

```
ALSA lib pcm_rate.c:1356:(snd_pcm_rate_open) Invalid type for rate converter
Button is pressed
[255 0 0 0 255 0 0 0 255]
Button is released
ALSA lib dlmisc.c:254:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_rate_.so
ALSA lib pcm_rate.c:1362:(snd_pcm_rate_open) Cannot find rate converter
```

Pillar1989, 2019-04-08:
Wouldn't it be better for you to ask the question at talkiepi? We didn't use talkiepi

schneekluth, 2019-04-08:
I did! I've been in contact with the developer for a week and we tried to fix it but couldn't figure it out what exactly was the problem. When using the old [0.3](https://github.com/respeaker/seeed-voicecard/archive/v3.0.tar.gz) version of the respeaker installer everything works fine. So I thought you might could have a look at it because the error only appears on the newest version of the installer script. I tried to create an additional pcm slave and defined it as type rate but could not get it to work so I thought maybe you could have a look at it.

closed 2019-04-16 by Pillar1989