respeaker/seeed-voicecard #149: Your kernel headers for kernel 4.19.34+ cannot be found at...

boggiano, 2019-04-11:
Hello,
I get this output when I try to install seeed-voicecard on my Rpi (Raspian 9 - 4.14.98-v7+)

```
DKMS: build completed.
Error! echo
Your kernel headers for kernel 4.19.34+ cannot be found at
/lib/modules/4.19.34+/build or /lib/modules/4.19.34+/source.
Error! echo
Your kernel headers for kernel 4.19.34-v7+ cannot be found at
/lib/modules/4.19.34-v7+/build or /lib/modules/4.19.34-v7+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 3fb4202] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.

```
If I try:

> sudo apt-get install raspberrypi-kernel-headers

I get:
`raspberrypi-kernel-headers is already the newest version (1.20190401-1).
`
Thanks

Pillar1989, 2019-04-13:
@boggiano #150

turmary, 2019-04-15:
Hi @boggiano 
We tried to fix this issue.
Please install again use the latest code.

boggiano, 2019-04-16:
Hello @turmary 
it worked!
Thank you!!

closed 2019-04-16 by boggiano