respeaker/seeed-voicecard #150: after a firmware and kernel update, 2mic not listed in aplay and arecord

maremoto, 2019-04-11:
Hi,

I had a 2-mic-hat working with raspberry pi 3B and raspbian 4.14.

Then I had to update the firmware and kernel distro 4.19.

Then I followed the steps in http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/ for installation but the aplay -l and arecord -l  do not show the card anymore.

Installation logs, no apparent troubles:

$ sudo apt-get update

Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Hit:3 https://raspbian.snips.ai/stretch stable InRelease
Reading package lists... Done
++ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

$ git clone https://github.com/respeaker/seeed-voicecard.git

Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 665 (delta 9), reused 23 (delta 9), pack-reused 642
Receiving objects: 100% (665/665), 1.31 MiB | 1.86 MiB/s, done.
Resolving deltas: 100% (408/408), done.

$ cd seeed-voicecard

$ sudo ./install.sh

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://raspbian.snips.ai/stretch stable InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20190401-1).
raspberrypi-kernel-headers is already the newest version (1.20190401-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u4).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.34+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.34+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.34+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.34+/kernel/sound/soc/bcm/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.34-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.34-v7+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.34-v7+/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.34-v7+/kernel/sound/soc/bcm/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.34+ -C /lib/modules/4.19.34+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....
cleaning build area...

DKMS: build completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.34-v7+ -C /lib/modules/4.19.34-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area...

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.34+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.34+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.34+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.

snd-soc-wm8960:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.34-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.34-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.34-v7+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 5fc4c3f] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

$ sudo reboot

maremoto, 2019-04-11:
Now the card is listed, because I added this (not in the installation guide):

sudo nano /boot/config.txt
   dtoverlay=seed-2mic-voicecard

But still does not work properly, I try to set the card as default mic/speaker but the /etc/asound.conf is erased in every reboot ¿¿??

Thanks,
Mar

turmary, 2019-04-15:
Hi @maremoto 
Our driver doesn't support kernel updated by rpi-update cause no coresspoding kernel-headers.

Please use below command to revert kernel version.
`sudo apt-get -y --reinstall install raspberrypi-kernel`

After then, try install again.

We updated the install.sh to warn this condition.

maremoto, 2019-04-15:
Hi,

Thanks a lot for answering, the problem is that other (non related with respeaker hat) software is requesting the firmware update, I will try to find out how to combine them.

Regards,
Mar

closed 2019-04-29 by Pillar1989