respeaker/seeed-voicecard #151: Not a issue, only adjust volume commands

Pengree, 2019-04-15:
Hello seeed guys,

I bought a Respeaker-2mic. Its hardware information is as followings:
```
pi@raspberry:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

My question is that what's the commands to adjust volume? (_**amixer sset**_ or other commands) I am going to use a application or a voice assistant to control the speaker volume.

Thanks!

Pillar1989, 2019-04-16:
To set the Playback or Capture volume use:
```
# Gets a list of simple mixer controls
$ amixer scontrols 
Then set it to the desired volume, as an example
$ amixer sset 'The Playback scontrols..... '  50%
```

Pengree, 2019-04-19:
amixer sset 'Playback' xx% for 3.5mm Audio Jack;
amixer sset 'Speaker' yy% for JST Speaker Connectors.

Thanks!

closed 2019-04-19 by Pengree