respeaker/seeed-voicecard #153: How to setup Hifiberry and Respeaker 4-mic I2C connection

tproulund, 2019-04-23:
Hallo

So I've seen this issued raised twice without much luck, but I find it a bit strange that shouldnt be a way to achieve this and a bit annoying since the combination of a Respeaker 4-mic and HiFiberry is such a logical choice for DIY Smart Speaker

you show that you have an external I2C port on the reSpeaker, is it possible to make the daisy-chain via that port? 
or what solutions can you think of to try? could it be possible to change the dsp_a format to a proper i2c?

I am really just looking for a solution that allows me to use both these nice products!

Pillar1989, 2019-04-29:
I have not tried to do this , you may be able to go to HiFiberry where to ask

tproulund, 2019-04-29:
Hi @Pillar1989 

But you have dealt with this previously: https://github.com/respeaker/seeed-voicecard/issues/43  &  https://github.com/respeaker/seeed-voicecard/issues/81

So now I am asking, if we can try to be more solution minded rather than dismissing the idea, so is it possible for you to look into how to pair up your product with a hifiberry?

baorepo, 2019-04-29:
I think this issue is very clear on this topic. #43

tproulund, 2019-04-30:
So you are saying there is absolutely no way of coming up with a working working solution and that it is impossible to pair up your device with any sort of I2C connection?

Pillar1989, 2019-05-08:
At least I can't help it

closed 2019-05-20 by Pillar1989