respeaker/seeed-voicecard #154: Respeaker 4 mics array and Linux USB Audio Gadget

letourng, 2019-04-26:
Hello, 

I'm trying to use respeaker mic array v1 with raspberry pi zero in USB mode. I would like to capture audio samples with the respeaker in 4 channels at 44100khz and send audio samples through USB. To do the job, I used USB audio gadget driver in the kernel on raspberry pi zero.

I used the following command:

arecord -Dsysdefault:CARD=seeed4micvoicec -c 4 -r 44100 -t raw -f S32_LE | aplay sysdefault:CARD=UAC1Gadget -f S32_LE -c 4 -r 44100
 
Results: Overrun with alsa

I also used the alsaloop command in root to have more priority on the thread but there're still to have overruns.

So some audio packet are missed in the audio file receive to usb.

Do you have some ideas to improve that?

Best regards

Pillar1989, 2019-04-29:
First of all, we must ensure that UAC1Gadget is stable. It has been tried by other users before, and it is very unstable.

letourng, 2019-04-29:
Ok so there are non solutions to use respeaker in USB mode?

baorepo, 2019-04-29:
To solve the stability problem of UAC1Gadget first, the main technical difficulty lies not in the seeed-voicecard side.

letourng, 2019-04-29:
Ok,

you said your team has been worked on UAC1gadget. What is the main issue in linux UACGadget side? 
Do you have any idea ?

Pillar1989, 2019-04-30:
We didn't analyze it in depth. We just did the test. The function we wanted was replaced by https://www.seeedstudio.com/ReSpeaker-Mic-Array-v2-0.html .

letourng, 2019-04-30:
Ok, 

Respeaker v2 supported 44Khz and 48khz 4 channels ?

jerryyip, 2019-05-08:
@letourng ReSpeaker Mic Array v2 supports only 48KHz and 16KHz by [different firmwares](https://github.com/respeaker/usb_4_mic_array), and it can output 6 channels(1 processed audio, 4 raw microphone audio, 1 reference channel audio).

closed 2019-05-20 by Pillar1989