respeaker/seeed-voicecard #156: Setup on arm archlinux, raspi zero w

idreamerhx, 2019-05-08:
I got this error:

[root@seeed-voicecard-20190509]# ./install.sh 
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
warning: git-2.21.0-1 is up to date -- skipping
warning: gcc-8.2.1+20181127-1 is up to date -- skipping
warning: automake-1.16.1-1 is up to date -- skipping
warning: make-4.2.1-3 is up to date -- skipping
warning: dkms-2.5-4 is up to date -- skipping
warning: linux-raspberrypi-headers-4.19.37-2 is up to date -- skipping
warning: i2c-tools-4.1-1 is up to date -- skipping
:: Starting full system upgrade...
 there is nothing to do
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.37+ cannot be found at
/usr/lib/modules/4.19.37+/build or /usr/lib/modules/4.19.37+/source.
Error! echo
Your kernel headers for kernel 4.19.37-v7+ cannot be found at
/usr/lib/modules/4.19.37-v7+/build or /usr/lib/modules/4.19.37-v7+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 77b1c2d] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service -> /usr/lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

[root@seeed-voicecard-20190509]# ls /usr/lib/modules/
4.14.73-1-ARCH/                4.19.37-2-ARCH/                extramodules-4.19-raspberrypi/ 
[root@ seeed-voicecard-20190509]# uname -a
Linux 4.19.37-2-ARCH #1 SMP PREEMPT Fri May 3 01:14:02 UTC 2019 armv6l GNU/Linux

Pillar1989, 2019-05-09:
Can't the sound card be detected after restarting the device?

idreamerhx, 2019-05-20:
Thanks for your reply!

[root@xiaoyuzai ~]# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
[root@xiaoyuzai ~]# ll /^C
[root@xiaoyuzai ~]# ^C
[root@xiaoyuzai ~]# ll  /usr/lib/modules
total 8
lrwxrwxrwx 1 root root   14 May 20 16:44 4.19.42+ -> 4.19.42-1-ARCH
drwxr-xr-x 4 root root 4096 May 20 16:49 4.19.42-1-ARCH
lrwxrwxrwx 1 root root   14 May 20 16:43 4.19.42-v7+ -> 4.19.42-1-ARCH
drwxr-xr-x 2 root root 4096 May 20 16:31 extramodules-4.19-raspberrypi

Pillar1989, 2019-05-20:
Duplicate of #159

closed 2019-05-20 by Pillar1989

idreamerhx, 2019-05-20:
I update this code to latest and update the system.

In /usr/lib/modules, I create 2 symbol links because it build (using install.sh) with some errors.

There is some additional infos:

[root@xiaoyuzai ~]# dmesg  | grep -i see  
[   28.288048] audit: type=1130 audit(1558342227.110:22): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=seeed-voicecard comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'