respeaker/seeed-voicecard #159: kernel 4.19.42-v7 expired seeed-voicecard

emilius3m, 2019-05-18:
after a kernel upgrade i have installed the kernel module following your guide but the only message that i view after the reboot is 

```
asoc-simple-card soc:sound: asoc_simple_card_dai_link_of: Can't find simple-audio-card,cpu DT node
[   48.276802] asoc-simple-card soc:sound: parse error -22
[   48.276848] asoc-simple-card: probe of soc:sound failed with error -22
```
```

root@pbx:~# uname -ar
Linux pbx 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
```

ABCarnage, 2019-05-18:
I've got the same problem. Got an hour and a half to try to fix this.

Daenara, 2019-05-18:
Same here but I wasn't even doing a kernel upgrade. I just got my respeaker 4 yesterday and I got around to playing with it an hour ago but so far I did an hour of reinstalling the drivers, trying to set it up via snips sam tool or manually, working through old threads looking for clues (there was one that suggested a kernel revert but I tried, it did something and then didn't change anything at all.) Kinda disappointing right now because everything I read before buying one was that it  will run basically without problems after installing the drivers and that snips can even set it up automatically. I hope someone figures out how to fix (or at least get around) this issue soon.

Daenara
(hopeful that I get to play with my board soon)

cornim, 2019-05-18:
Yep, same issue here as well. Possibly relevant lines from dmesg are:

`[   18.032716]  ac10x i2c_id number: 0
[   18.032722]  ac10x data protocol: 0
[   18.158971]  ac10x codec count  : 1
[   18.159333] seeed-voicecard soc:sound: seeed_voice_card_dai_link_of: Can't find seeed-voice-card,codec DT node
[   18.159340] seeed-voicecard soc:sound: parse error -22
[   18.159354] seeed-voicecard: probe of soc:sound failed with error -22
`

Pillar1989, 2019-05-19:
This should be caused by kernel update 4.19, we are trying to fix it, and we will release the corresponding patch soon.

Pillar1989, 2019-05-19:
If we want to keep the kernel in version 4.14, do you accept it? In other words, as long as you use the install.sh script, the kernel rollback operation will be executed.

Daenara, 2019-05-19:
Will it roll back even when the system came with a newer kernel pre-installed (because I don't know how to get it down from there)? Keeping an older kernel will be an issue in the future because other things might need a newer one but I think for now ppl will settle for a working mic array

emilius3m, 2019-05-19:
if you want you can declare the product compatible only with kernel 4.14 and adopt this solution. but I don't think that in the long term it can be adopted because every future update will reinstall the latest version

cornim, 2019-05-19:
Rolling back to 4.14 might work as a short term solution until you get it to work with 4.19. Keeping it at 4.14 permanently would certainly not work for me.

turmary, 2019-05-20:
hi all,
In RPi Linux v4.19, the root error
```
[   13.640081] ac10x-codec 1-001a: prop pinctrl-0 index 0 invalid phandle
[   13.651109] ac10x-codec: probe of 1-001a failed with error -22
```
result in a sound card probe failing.

I think it's a bug of device tree part in RPi kernel v4.19.

thangnd85, 2019-05-20:
Temp revert  kernel and kernel-headers to 4.14, then mark hold upgrade may help you. But it's not solution.

turmary, 2019-05-21:
Hi all,
We tried to fix this problem by inserting overlay into /boot/config.txt.
Please update code, uninstall then install again.

ABCarnage, 2019-05-21:
> Hi all,
> We tried to fix this problem by inserting overlay into /boot/config.txt instead of reverting kernel version to v4.14.
> Please update code, uninstall then install again.

Can you please give us more details? What overlay should I insert?

ABCarnage, 2019-05-21:
Ok, I tested it, it works but now there is no "/etc/asound.conf" file, but I know where to find it in this repo.
Congrats @turmary

cornim, 2019-05-21:
Just tested in on a clean install of strech lite 4.19.
Also ran into the mission asound.conf issue but after manually copying it, everything seems to work fine.

snail59, 2019-05-21:
Hello,
I think that the /etc/asound.conf file is not missing but is deleted; at least, that is the case for me.
seeed-voicecard.service ( systemd service file) calls /usr/bin/seeed-voicecard when the system starts or reboots.
This file contains, on line 87 and 88
```
rm /etc/asound.conf
rm /var/lib/alsa/asound.state
```
So both files are deleted.
Then, it tries to guess which card is installed (mine is respeaker 2-mic)
```
if [ "x${is_1a}" == "x1a" ] && [ "x${is_35}" == "x" ] ; then
    echo "install 2mic"
    overlay=seeed-2mic-voicecard
    rm /etc/asound.conf
    rm /var/lib/alsa/asound.state
    echo "create 2mic asound configure file"
    ln -s /etc/voicecard/asound_2mic.conf /etc/asound.conf

    echo "create 2mic asound status file"
    ln -s /etc/voicecard/wm8960_asound.state /var/lib/alsa/asound.state
fi
```

where is_1a the result of this command:
```
is_1a=$(i2cdetect -y  1 0x1a 0x1a | grep 1a | awk '{print $2}')
```
I guess there is a problem with the output of the i2cdetect command  :
```
# i2cdetect -y  1 0x1a 0x1a
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                               UU                
20:                                                 
30:                                                 
40:                                                 
50:                                                 
60:                                                 
70: 
```
I suppose that the ouptut should contain "1a" somewhere.
That is not the case so the "ln" commands are not executed, leaving the system with neither /etc/asound.conf nor /var/lib/alsa/asound.state file.

laguerreche, 2019-05-21:
I ran the _"./install.sh"_ script with the commit [0ea5d2b](https://github.com/respeaker/seeed-voicecard/commit/0ea5d2bd98c47e2068498d44516ab6316394e84d) but my 2mic-voicecard doesn't seem to work again.

I have the _/etc/asound.conf_ file but nor Pulseaudio neither _"arecord -l"_ find microphones:
```
$ pactl list short sinks
0	alsa_output.platform-3f902000.hdmi.iec958-stereo	module-alsa-card.c	s16le 2ch 44100Hz	SUSPENDED
1	alsa_output.usb-Generic_USB2.0_Device_20130100ph0-00.analog-stereo	module-alsa-card.c	s16le 2ch 48000Hz	SUSPENDED
2	alsa_output.platform-soc_audio.analog-stereo	module-alsa-card.c	s16le 2ch 44100Hz	SUSPENDED
```

for arecord, I get an empty result.

The seed-voice modules seem to be loaded:
```
$ lsmod | grep see
snd_soc_seeed_voicecard    16384  1 snd_soc_ac108
snd_soc_simple_card_utils    16384  2 snd_soc_seeed_voicecard,snd_soc_simple_card
snd_soc_core          192512  7 vc4,snd_soc_seeed_voicecard,snd_soc_simple_card_utils,snd_soc_bcm2835_i2s,snd_soc_ac108,snd_soc_wm8960,snd_soc_simple_card
```

I get weird results with systemd when requesting status of the seed service:
```
$ sudo systemctl status seeed-voicecard
● seeed-voicecard.service - Seeed Voicecard service
   Loaded: loaded (/lib/systemd/system/seeed-voicecard.service; enabled; vendor preset: enabled)
   Active: active (exited) since Tue 2019-05-21 20:30:23 CEST; 1h 8min ago
  Process: 445 ExecStart=/usr/bin/seeed-voicecard (code=exited, status=0/SUCCESS)
 Main PID: 445 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/seeed-voicecard.service

mai 21 20:30:18 raspberrypi su[1038]: Successful su for root by root
mai 21 20:30:18 raspberrypi su[1038]: + ??? root:root
mai 21 20:30:18 raspberrypi su[1038]: pam_unix(su:session): session opened for user root by (uid=0)
mai 21 20:30:19 raspberrypi su[1130]: Successful su for root by root
mai 21 20:30:19 raspberrypi su[1130]: + ??? root:root
mai 21 20:30:19 raspberrypi su[1130]: pam_unix(su:session): session opened for user root by (uid=0)
mai 21 20:30:19 raspberrypi su[1156]: Successful su for root by root
mai 21 20:30:19 raspberrypi su[1156]: + ??? root:root
mai 21 20:30:20 raspberrypi su[1156]: pam_unix(su:session): session opened for user root by (uid=0)
mai 21 20:30:23 raspberrypi systemd[1]: Started Seeed Voicecard service.
```

Finally, in the output of dmesg, I obtain the following scaring messages:
```
[   16.351275] asoc-simple-card soc:sound: asoc_simple_card_dai_link_of: Can't find simple-audio-card,cpu DT node
[   16.351293] asoc-simple-card soc:sound: parse error -22
[   16.351325] asoc-simple-card: probe of soc:sound failed with error -22
```

OttPeterR, 2019-05-22:
I'm new to the respeaker world, I just bought the 4-mic array yesterday and tried it out with kernel 4.19.42-v7+ and it looks like I have the same issue after installing the drivers. I did run the pixel tests though so I know my device isn't broken. My system is showing everything @laguerreche post above shows.

turmary, 2019-05-22:
Hi all,
I tried to fix the problem missing asound.conf.
[346599](https://github.com/respeaker/seeed-voicecard/commit/da5f90473b6fe30b4857c6704bd9af3f2e346599)
Please uninstall, reboot(poweroff),insert sound card, boot, install, then reboot again.

OttPeterR, 2019-05-22:
That fixed it for me, thanks!

closed 2019-05-30 by Pillar1989

laguerreche, 2019-05-31:
Sorry but that still doesn't work for me.

_switch off_,
I removed my card,
_boot_,
I uninstalled the drivers (with the uninstall script),
_reboot_,
I inserted the card,
_reboot_,
I reinstalled the drivers (with the install script),
and now _/etc/asound.conf_ and _/var/lib/alsa/asound.state_ don't exist.

Any idea? I have a 2mics card.

laguerreche, 2019-05-31:
I am not sure that the _seeed-voicecard_ script gets what it expects:
```
$ i2cdetect -y  1 0x1a 0x1a
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                               UU                
20:                                                 
30:                                                 
40:                                                 
50:                                                 
60:                                                 
70:                                                 
```

So, no output with that: `is_1a=$(i2cdetect -y  1 0x1a 0x1a | grep 1a | awk '{print $2}')`.

snail59, 2019-05-31:
hello,
the last commit ( git log) is da5f90473b6fe30b4857c6704bd9af3f2e346599, in which the test is now
`is_1a=$(i2cdetect -y  1 0x1a 0x1a | egrep "(1a|UU)" | awk '{print $2}')`. Notice "(1a|UU)" instead of grep 1a only.
You should maybe run git update.

By the way @turmary  , egrep has been deprecated in favor of grep -E for a few years now.

laguerreche, 2019-06-02:
Damn, I failed to update the git repo before running again the scripts...
Now, I'm again able to record sound.

The output of `systemctl status seeed-voicecard` is still strange:
```
$ sudo systemctl status seeed-voicecard
● seeed-voicecard.service - Seeed Voicecard service
   Loaded: loaded (/lib/systemd/system/seeed-voicecard.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2019-06-01 17:24:11 CEST; 22h ago
  Process: 451 ExecStart=/usr/bin/seeed-voicecard (code=exited, status=0/SUCCESS)
 Main PID: 451 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/seeed-voicecard.service

juin 01 17:24:07 raspberrypi su[670]: Successful su for root by root
juin 01 17:24:07 raspberrypi su[670]: + ??? root:root
juin 01 17:24:07 raspberrypi su[670]: pam_unix(su:session): session opened for user root by (uid=0)
juin 01 17:24:07 raspberrypi su[753]: Successful su for root by root
juin 01 17:24:07 raspberrypi su[753]: + ??? root:root
juin 01 17:24:07 raspberrypi su[753]: pam_unix(su:session): session opened for user root by (uid=0)
juin 01 17:24:08 raspberrypi su[800]: Successful su for root by root
juin 01 17:24:08 raspberrypi su[800]: + ??? root:root
juin 01 17:24:08 raspberrypi su[800]: pam_unix(su:session): session opened for user root by (uid=0)
juin 01 17:24:11 raspberrypi systemd[1]: Started Seeed Voicecard service.
```

However, I'm trying to use Snips but I'm facing two opposite ideas:
- Snips expects a specific configuration in _/etc/asound.conf_, see https://docs.snips.ai/articles/raspberrypi/manual-setup,
- your installation scripts overwrite the file _/etc/asound.conf_.

So, what am I supposed to do?

In addition, I'm also surprised to see "dmix" in _/etc/asound.conf_ because Pulseaudio -- that I use -- was supposed to replace Dmix. Is your configuration appropriate for Pulseaudio?

Pillar1989, 2019-06-03:
Pulseaudio  has not been tested

samuelbles07, 2019-07-02:
any update on this?