respeaker/seeed-voicecard #160: Respeaker 2 Mic on PI Zero (strech) not found

opl-git, 2019-05-18:
Hello,
I am using a raspberry pi zero 1.1 W running the lastet debian strech release (2019-04-08).
Before the installation I did the usual install stuff (apt-get update/upgrade, expand filesystem, change localization to germany).
I tried to install the Respeaker 2 Mics Version like described here:
`git clone https://github.com/respeaker/seeed-voicecard
cd seeed-voicecard
sudo ./install.sh 
sudo reboot`

It looks like everything went well, but after the reboot I can`t find the respeakercard with _aplay -l_ or _arecord - l_

Is there something I forgot?
Hope someone may help me :)

Pillar1989, 2019-05-19:
Duplicate of #160

Pillar1989, 2019-05-19:
#160

closed 2019-05-19 by Pillar1989

opl-git, 2019-05-19:
I am confused... this is a duplicate of my own question?
You closed it, because it is a duplicate of #160 which is my issue...

Pillar1989, 2019-05-19:
I am really sorry, your question should be duplicated with 159. I filled in the wrong number.

opl-git, 2019-05-19:
I see. 😊
So it is because of the new kernel.
I am looking forward for your patch, thank you very much for your efforts.