respeaker/seeed-voicecard #162: install respeaker on a specific linux version (4.14.114)

jonspieg, 2019-05-28:
Hi,

When I install my respeaker according to the guides everything seems to work fine, however my linux kernel version changes to 4.14.98. This is very problematic for me because I have some custom modules installed on a very specific kernel version (4.14.114) which I've built myself, I simply cannot use any other version.

Is there a way of installing it on the current kernel version without updating it?

Thanks,
Jonathan

cpoisson, 2019-05-28:
I suppose you need to checkout a previous version of the driver.

probably before this commit https://github.com/respeaker/seeed-voicecard/commit/b98fce84e0e45eaa85ef08c4b5dfe962b49a5149

adrianhoodmech, 2019-05-29:
`#!/bin/bash

sudo apt-get update
sudo apt-get upgrade
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
git checkout c7026a40487a6dc751c93d3d566a4e0b5fd93442
git pull
sudo ./install.sh
sudo reboot`

adrianhoodmech, 2019-05-29:
#!/bin/bash

sudo apt-get update
sudo apt-get upgrade
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
git checkout c7026a40487a6dc751c93d3d566a4e0b5fd93442
git pull
sudo ./install.sh
'sudo reboot'

adrianhoodmech, 2019-05-29:
Try rolling back to the previous version of install.sh in install_respeaker.sh using git checkout, there seems to be a problem with the latest version.

baorepo, 2019-05-30:
Install respeaker on Linux 4.14 version method:
https://github.com/respeaker/seeed-voicecard/blob/0ea5d2bd98c47e2068498d44516ab6316394e84d/install.sh#L75-L82

Modify it to the following code

```
 which apt &>/dev/null 
 if [[ $? -eq 0 ]]; then 
   apt update -y 
   wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20190401-1_armhf.deb -O /tmp/raspberrypi-kernel-headers_1.20190401-1_armhf.deb 
   wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190401-1_armhf.deb -O /tmp/raspberrypi-kernel_1.20190401-1_armhf.deb
   dpkg -i  /tmp/raspberrypi-kernel-headers_1.20190401-1_armhf.deb
   dpkg -i  /tmp/raspberrypi-kernel_1.20190401-1_armhf.deb
   apt-get -y install dkms git i2c-tools libasound2-plugins 
   # rpi-update checker 
   check_kernel_headers 
fi 
```

closed 2019-06-04 by Pillar1989

cpoisson, 2019-06-11:
@Pillar1989 

A small suggestion.

IMHO, It could be a good to idea to add version tag or maintenance branch to your repository to deal with those kind of issues.