respeaker/seeed-voicecard #163: Raspberry crashes when activating LEDs

cburghardt, 2019-06-01:
I have a Raspberry Zero WH with Respeaker 4. The micro works just fine.
Once I try to activate any LED the pi crashes and reboots. I have tested with the pixel ring as well as 4mics_hat (pixel demo).
What can I do?

Pillar1989, 2019-06-03:
You mean it crashed when you tested pixel ring? Can you make your damn issue detail?

cburghardt, 2019-06-03:
Yes, I run either the pixel ring or the 4mics_hat, they freeze, I loose the connection to the PI and it reboots. There are no messages in either /var/log/syslog, messages or kern.log.
In order to rule out an issue with the gpiozero library I have also tested the snips-respeaker-skill which has the same issue.
I have stripped down the pixels.py to the following line which causes the reboot:
self.power = LED(5)

So it already seems to crash when setting the pin 5.

Pillar1989, 2019-06-03:
https://github.com/RPi-Distro/python-gpiozero/issues 
Is it better to discuss your question here?

cburghardt, 2019-06-03:
As mentioned I can reproduce the issue also with the snips-respeaker-skill which doesn't use that library. It is an issue with the respeaker board but I would like to get an idea if there is a chance to debug what is going on.

Pillar1989, 2019-06-04:
@cburghardt  You can try if you don't use our board and see if there is a problem

cburghardt, 2019-06-04:
When I write "1" to gpio5 the PI crashes with the board. Without board there is no problem. I will send it back.

closed 2019-06-25 by Pillar1989

drivin, 2019-12-06:
> When I write "1" to gpio5 the PI crashes with the board. Without board there is no problem. I will send it back.

The same issue here. Did you solve this issue?

cburghardt, 2019-12-06:
Yes the replacement Pi did not have any issue.

Lars-L00t, 2021-04-19:
> 
> 
> > When I write "1" to gpio5 the PI crashes with the board. Without board there is no problem. I will send it back.
> 
> The same issue here. Did you solve this issue?

Anything new on this one? Besides replacing my Pi...