respeaker/seeed-voicecard #164: Still can't setup on archlinux raspi zero w

idreamerhx, 2019-06-02:
[root@xiaoyuzai seeed-voicecard-20190526]# ./install.sh 
pacman -Syu --needed git gcc automake make dkms linux-raspberrypi-headers i2c-tools

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.46-1-ARCH (armv6l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko.gz:
 - Uninstallation
   - Deleting from: /usr/lib/modules/4.19.46-1-ARCH/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko.gz:
 - Uninstallation
   - Deleting from: /usr/lib/modules/4.19.46-1-ARCH/kernel/sound/soc/codecs/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko.gz:
 - Uninstallation
   - Deleting from: /usr/lib/modules/4.19.46-1-ARCH/kernel/sound/soc/bcm/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod......

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.46+ cannot be found at
/usr/lib/modules/4.19.46+/build or /usr/lib/modules/4.19.46+/source.
Error! echo
Your kernel headers for kernel 4.19.46-v7+ cannot be found at
/usr/lib/modules/4.19.46-v7+/build or /usr/lib/modules/4.19.46-v7+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 1a4fdf0] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service -> /usr/lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

idreamerhx, 2019-06-02:
total 8
lrwxrwxrwx 1 root root   15 Jun  2 22:08 4.19.46+ -> 4.19.46-1-ARCH/
drwxr-xr-x 4 root root 4096 Jun  2 21:49 4.19.46-1-ARCH
lrwxrwxrwx 1 root root   15 Jun  2 22:09 4.19.46-v7+ -> 4.19.46-1-ARCH/
drwxr-xr-x 2 root root 4096 Jun  2 21:37 extramodules-4.19-raspberrypi
 pwd
/usr/lib/modules

idreamerhx, 2019-06-02:
I create links it works

Pillar1989, 2019-06-03:
So any other issue?

closed 2019-06-25 by Pillar1989