respeaker/seeed-voicecard #166: Seeed voicecard not properly initialized

norbusan, 2019-06-05:
Hi all,
it seems on latest raspbian I cannot install the seeed voicecard. Installation runs fine, but there are a few issues:
- the kernel `config.txt` gets entries
```
dtoverlay=i2s-mmap
dtparam=i2s=on
dtoverlay=seeed-2mic-voicecard
```
but it seems that `i2s-mmap` is gone:
```
$ sudo dtoverlay i2s-mmap
DTOVERLAY[error]: Failed to open '/boot/overlays/i2s-mmap.dtbo'
* Failed to read '/boot/overlays/i2s-mmap.dtbo'
$
```
Furthermore, then the seeed-2mic overlay cannot be loaded:
```
$ sudo dtoverlay seeed-2mic-voicecard
* Failed to apply overlay '4_seeed-2mic-voicecard' (kernel)
$
```
This is with kernel
```
Linux raspberrypi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
```
and the kernel package is raspberrypi-kernel 1.20190517-1. I checked the changes in the kernel packages and the associated firmware package, but couldn't spot anything strange.

- I guess as a consequence, loading the alsa state fails:
```
$ sudo alsactl restore
Found hardware: "bcm2835_alsa" "Broadcom Mixer" "" "" ""
Hardware is initialized using a generic method
alsactl: set_control:1325: failed to obtain info for control #6 (No such file or directory)
```

Could you please look into this and advice what to do?

norbusan, 2019-06-05:
I attach the log file created in `/var/log/seeed-voicecard.log`
[seeed-voicecard.log](https://github.com/respeaker/seeed-voicecard/files/3255270/seeed-voicecard.log)

Pillar1989, 2019-06-05:
dtoverlay is no longer functional on the latest kernel, and if you want to add dtbo files dynamically, you can only read and write config.txt files

norbusan, 2019-06-05:
Oh??? Is this a "bug" or a "feature" of the latest kernel? I thought dtoverlay is necessary to activate various functionalities for additional hats. I would be surprised if removing that functionality was on purpose... Do you @Pillar1989 have more information about this?

norbusan, 2019-06-05:
Ah, maybe you meant that loading is only possible via config.txt. But even there it does not work ...

Pillar1989, 2019-06-05:
It works. Our default install.sh script is used in this way.

norbusan, 2019-06-05:
Hmmm, I *am* running your install.sh script right from the git repository checkout ... the above was after running the script and rebooting.

wdavalil, 2019-06-10:
im haven a similar problem too

martinjuhasz, 2019-06-12:
just freshly installed latest raspbian and `apt-get upgrade`'d my device. cloned the repo and installed it. went fine but afterwards i get the same problems as above. also the device doesn't show up on alsa.
```
pi@snips-01:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@snips-01:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@snips-01:~ $ sudo dtoverlay i2s-mmap
DTOVERLAY[error]: Failed to open '/boot/overlays/i2s-mmap.dtbo'
* Failed to read '/boot/overlays/i2s-mmap.dtbo'
pi@snips-01:~ $
```

martinjuhasz, 2019-06-12:
Strange, i just used an older `2018-11-13-raspbian-stretch` image of raspbian, didn't upgrade it and went straight to install. after reboot i still don't see the playback or recording device listed.

any idea what is happening?

```
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh
sudo reboot
```

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
```

turmary, 2019-06-13:
@martinjuhasz  Please show me the log of below commands.
dmesg
cat /var/log/seeed-voicecard.log

martinjuhasz, 2019-06-13:
OK sure. Which commands should i have run before so the log makes sense?
What i now did:
```
aplay -l
arecord -l
sudo dtoverlay i2s-mmap
sudo dtoverlay seeed-2mic-voicecard
```

<details>
    <summary> dmesg</summary>

```
    [    0.000000] Booting Linux on physical CPU 0x0
    [    0.000000] Linux version 4.19.42+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1219 Tue May 14 21:16:38 BST 2019
    [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
    [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
    [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
    [    0.000000] Memory policy: Data cache writeback
    [    0.000000] cma: Reserved 8 MiB at 0x1b400000
    [    0.000000] On node 0 totalpages: 114688
    [    0.000000]   Normal zone: 1008 pages used for memmap
    [    0.000000]   Normal zone: 0 pages reserved
    [    0.000000]   Normal zone: 114688 pages, LIFO batch:31
    [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
    [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
    [    0.000000] pcpu-alloc: [0] 0
    [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
    [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:75:B2:4A vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=d532b144-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
    [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
    [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
    [    0.000000] Memory: 434484K/458752K available (6932K kernel code, 633K rwdata, 2068K rodata, 456K init, 797K bss, 16076K reserved, 8192K cma-reserved)
    [    0.000000] Virtual kernel memory layout:
                    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                    vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                    lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                    modules : 0xbf000000 - 0xc0000000   (  16 MB)
                        .text : 0x(ptrval) - 0x(ptrval)   (6934 kB)
                        .init : 0x(ptrval) - 0x(ptrval)   ( 456 kB)
                        .data : 0x(ptrval) - 0x(ptrval)   ( 634 kB)
                        .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
    [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
    [    0.000000] ftrace: allocating 25044 entries in 74 pages
    [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
    [    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
    [    0.000087] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
    [    0.000195] bcm2835: system timer (irq = 27)
    [    0.000793] Console: colour dummy device 80x30
    [    0.001431] console [tty1] enabled
    [    0.001513] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
    [    0.060374] pid_max: default: 32768 minimum: 301
    [    0.060970] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
    [    0.061030] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
    [    0.062498] CPU: Testing write buffer coherency: ok
    [    0.063887] Setting up static identity map for 0x8200 - 0x8238
    [    0.065347] devtmpfs: initialized
    [    0.076318] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
    [    0.076784] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
    [    0.076859] futex hash table entries: 256 (order: -1, 3072 bytes)
    [    0.078251] pinctrl core: initialized pinctrl subsystem
    [    0.079884] NET: Registered protocol family 16
    [    0.083209] DMA: preallocated 1024 KiB pool for atomic coherent allocations
    [    0.090376] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
    [    0.090553] hw-breakpoint: maximum watchpoint size is 4 bytes.
    [    0.090791] Serial: AMBA PL011 UART driver
    [    0.094386] bcm2835-mbox 2000b880.mailbox: mailbox enabled
    [    0.138062] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
    [    0.140631] SCSI subsystem initialized
    [    0.141155] usbcore: registered new interface driver usbfs
[... 316 lines cut ...]
    ++ ov=seeed-8mic-voicecard
    ++ grep -q -E '^dtoverlay=seeed-8mic-voicecard' /boot/config.txt
    ++ echo 1
    + '[' 1 -eq 0 ']'
    + '[' 1 -eq 0 ']'
    + '[' 1 -eq 0 ']'
    + '[' 1 -eq 1 ']'
    + sed /boot/config.txt -i -e 's/^dtoverlay=seeed-8mic-voicecard/#dtoverlay=seeed-8mic-voicecard/'
    + STATUS=disabled
    + echo 'create seeed-2mic-voicecard asound configure file'
    create seeed-2mic-voicecard asound configure file
    + ln -s /etc/voicecard/asound_2mic.conf /etc/asound.conf
    + echo 'create seeed-2mic-voicecard asound status file'
    create seeed-2mic-voicecard asound status file
    + ln -s /etc/voicecard/wm8960_asound.state /var/lib/alsa/asound.state
    + alsactl restore
    Found hardware: "bcm2835_alsa" "Broadcom Mixer" "" "" ""
    Hardware is initialized using a generic method
    alsactl: set_control:1325: failed to obtain info for control #6 (No such file or directory)
    + amixer cset numid=3 1
    numid=3,iface=MIXER,name='PCM Playback Route'
    ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
    : values=1
```
</details>

turmary, 2019-06-17:
hi @martinjuhasz 
Nothing need to do except reboot.

Below commands
```
sudo dtoverlay i2s-mmap
sudo dtoverlay seeed-2mic-voicecard
```
not functional.

dtbo is configed by /boot/config.txt and load by kernel while not above commands.

cydia2020, 2019-06-19:
Same issue here. seeed-voicecard.service is not able to start, cannot find i2s-mmap when trying to dtoverlay it.

turmary, 2019-06-19:
https://github.com/raspberrypi/linux/issues/2995#issue-451152007

turmary, 2019-06-19:
#https://github.com/raspberrypi/linux/issues/2995#issue-451152007

turmary, 2019-06-19:
#https://github.com/raspberrypi/linux/issues/2995

turmary, 2019-06-19:
@cydia2020 we don't need dtoverlay command in kernel version 4.19.x.

pelwell, 2019-06-19:
The i2s-mmap overlay has gone because the "mmap" mode is enabled by default since the 4.7 kernel.

martinjuhasz, 2019-06-20:
this seems to fail for everyone with the latest raspbian image, trying to install seeed-voicecard. i'm using a raspberry pi zero w.

@cydia2020 @wdavalil @norbusan on which device do you get this problem?

norbusan, 2019-06-20:
RPi 3b+, latest raspbian image

cydia2020, 2019-06-20:
RPi 3B+, latest raspbian

cyberius0, 2019-06-22:
RPi3+, did apt upgrade, respeaker is not listed in arecord -L
I really thought it is damaged till I found this open issue. Did anybody solve this?

DigitalElise, 2019-06-24:
Same here, RPi 3B+, latest raspbian.

Raspbian Buster Lite
Minimal image based on Debian Buster
Version:June 2019
Release date:2019-06-20
Kernel version:4.19

norbusan, 2019-06-24:
@DigitalElise Hmm, isn't that already the buster based Raspbian? There the compilation of the reseed card doesn't work, at least in my case, since the `fixdep` utility in the raspberrypi-kernel-headers:armhf packages is a x86_64 binary ...

thangnd85, 2019-06-24:
Same here, Raspbian Buster
Kernel 4.19.50
Trying to revert to 4.19.42 
`wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190517-1_armhf.deb
sudo dpkg -i raspberrypi-kernel_1.20190517-1_armhf.deb
cd ~
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20190517-1_armhf.deb
sudo dpkg -i raspberrypi-kernel-headers_1.20190517-1_armhf.deb
sudo apt-mark hold raspberrypi-kernel raspberrypi-kernel-headers`
Ofcourse, it work. But hope that we can find a better solution

Pillar1989, 2019-06-25:
@thangnd85 You are right. We will solve it soon.

norbusan, 2019-06-25:
The bug with fixdep I reported to raspbian and it was fixed over night, see https://github.com/raspberrypi/firmware/issues/1155

Pillar1989, 2019-06-25:
@norbusan Thank you so much for providing such important information. It seems that the problem is not on our side. All we can do is wait.

norbusan, 2019-06-26:
@Pillar1989 I can confirm that with the latest state of the Raspbian mirrors I can compile, install, and after reboot the card is properly initialized.

Thanks a lot. I think we can close this issue.

closed 2019-06-26 by Pillar1989

martinjuhasz, 2019-06-27:
cool! Thanks to all for keeping an eye on this and providing information.

bachoo786, 2019-07-12:
I am still facing the same issue as described in the original OP. Do we have a solution yet? Thanks