respeaker/seeed-voicecard #168: Raspberry Pi 4 Support

bencevans, 2019-07-09:
Hi

Trying to setup the ReSpeaker 4 mic array hat on a Raspberry Pi 4 however running into issues in that systemd-modules-load.service fails to load and unable to capture from device using arecord...

Installation log:

```
$ sudo ./install.sh 
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]                       
Hit:3 http://download.zerotier.com/debian/buster buster InRelease                                          
Hit:4 https://deb.nodesource.com/node_12.x buster InRelease               
Get:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.0 MB in 14s (950 kB/s)                                                                                                                                      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20190620+1-1).
raspberrypi-kernel-headers is already the newest version (1.20190620+1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.50+ -C /lib/modules/4.19.50+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.50+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.50+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.50+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.50+/kernel/sound/soc/codecs/snd-soc-ac108.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.50+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.50+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.50+/kernel/sound/soc/bcm/snd-soc-seeed-voicecard.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.50+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.50+/kernel/sound/soc/bcm/

depmod...
Job for systemd-modules-load.service failed because the control process exited with error code.
See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.

DKMS: install completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.50-v7+ -C /lib/modules/4.19.50-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.50-v7+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.50-v7+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.50-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.50-v7+/kernel/sound/soc/codecs/snd-soc-ac108.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.50-v7+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.50-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.50-v7+/kernel/sound/soc/bcm/snd-soc-seeed-voicecard.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.50-v7+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.50-v7+/kernel/sound/soc/bcm/

depmod....
Job for systemd-modules-load.service failed because the control process exited with error code.
See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.

DKMS: install completed.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) af1a831] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

... Followed by reboot...

```
sudo reboot
```

systemd-modules-load failing to load

```
$ systemctl status systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-07-09 18:50:38 BST; 1min 28s ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 299 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
 Main PID: 299 (code=exited, status=1/FAILURE)

Jul 09 18:50:38 raspberrypi systemd[1]: Starting Load Kernel Modules...
Jul 09 18:50:38 raspberrypi systemd-modules-load[299]: Failed to find module 'snd-soc-seeed-voicecard'
Jul 09 18:50:38 raspberrypi systemd-modules-load[299]: Failed to find module 'snd-soc-ac108'
Jul 09 18:50:38 raspberrypi systemd-modules-load[299]: Failed to find module 'snd-soc-wm8960'
Jul 09 18:50:38 raspberrypi systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Jul 09 18:50:38 raspberrypi systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Jul 09 18:50:38 raspberrypi systemd[1]: Failed to start Load Kernel Modules.
```

arecord list

```
$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
playback
ac108
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output
```

attempting to record

```
$ arecord -Dac108 -f S32_LE -r 16000 -c 4 a.wav
ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
arecord: main:828: audio open error: No such device
```

uname

```
$ uname -a
Linux raspberrypi 4.19.50-v7l+ #895 SMP Thu Jun 20 16:03:42 BST 2019 armv7l GNU/Linux
```

Any help would be much appreciated. Happy to send further information!

Pillar1989, 2019-07-10:
Duplicate of #166

Pillar1989, 2019-07-10:
the problem is not on our side. All we can do is wait.

norbusan, 2019-07-10:
@Pillar1989 I don't think this is the same error as #166 This bug is already fixed and so should not show up any more.

The RPi4 (new hardware!) exhibits a different bug, and I have seen this myself (but currently I am out of my home so I cannot investigate it further).

The compilation etc all works well, but somehow the module isn't loaded properly.

When I am back I will send a debug log and everything I can find.

So I suggest for now to un-mark this as duplicated and reopen

Pillar1989, 2019-07-10:
@norbusan This is because 4.19.50-v7+ headers does not match with 4.19.50 image. When you feedback the problem, the raspberry PI official deal, they put the version did not match.

bencevans, 2019-07-10:
I'm seeing the following in seeed-voicecard.log:

```
Hardware is initialized using a generic method
alsactl: set_control:1325: failed to obtain info for control #6 (No such file or directory)
```

Attached whole log in case useful [seeed-voicecard.log](https://github.com/respeaker/seeed-voicecard/files/3376903/seeed-voicecard.log)

a375702, 2019-07-17:
I see the same failure on 4.19.57.

Linux picroft 4.19.57-v7l+ #1244 SMP Thu Jul 4 18:48:07 BST 2019 armv7l GNU/Linux

$systemctl status systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-07-17 18:00:09 EDT; 5min ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 296 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
 Main PID: 296 (code=exited, status=1/FAILURE)

Jul 17 18:00:09 picroft systemd[1]: Starting Load Kernel Modules...
Jul 17 18:00:09 picroft systemd-modules-load[296]: Failed to find module 'snd-soc-seeed-voicecard'
Jul 17 18:00:09 picroft systemd-modules-load[296]: Failed to find module 'snd-soc-ac108'
Jul 17 18:00:09 picroft systemd-modules-load[296]: Failed to find module 'snd-soc-wm8960'
Jul 17 18:00:09 picroft systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Jul 17 18:00:09 picroft systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Jul 17 18:00:09 picroft systemd[1]: Failed to start Load Kernel Modules.

nedialkom, 2019-07-18:
I'm Experiencing the same issue with Raspberry Pi 4:

pi@raspberrypi:~ $ systemctl status systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; ven
   Active: failed (Result: exit-code) since Thu 2019-07-18 18:33:09 EEST; 3min 3
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 307 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=
 Main PID: 307 (code=exited, status=1/FAILURE)

Jul 18 18:33:09 raspberrypi systemd[1]: Starting Load Kernel Modules...
Jul 18 18:33:09 raspberrypi systemd-modules-load[307]: Failed to find module 'sn
Jul 18 18:33:09 raspberrypi systemd-modules-load[307]: Failed to find module 'sn
Jul 18 18:33:09 raspberrypi systemd-modules-load[307]: Failed to find module 'sn
Jul 18 18:33:09 raspberrypi systemd[1]: systemd-modules-load.service: Main proce
Jul 18 18:33:09 raspberrypi systemd[1]: systemd-modules-load.service: Failed wit
Jul 18 18:33:09 raspberrypi systemd[1]: Failed to start Load Kernel Modules.

Hope you will find soon the solution!

Pillar1989, 2019-07-25:
This problem should be solved. I have tested it and there is no problem
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

norbusan, 2019-07-26:
@Pillar1989 thanks a lot for the information. When I am back home and have access to device I will try it out, too. Was this a change in your repo or in the upstream (raspbian) kernel sources/packaging?

Again, thanks a lot!

bencevans, 2019-07-26:
I can confirm it's all working! I had to wipe the SD card and install from scratch as uninstalling/installing by itself didn't quite do the job. Thanks all!

closed 2019-07-26 by bencevans

ismailryt, 2020-05-07:
@bencevans hello i now is quite a while after you post this , but this mean you just reinstall all of them and install it again?

bencevans, 2020-05-07:
@ismailryt from what I recall, I still had issues after uninstalling and reinstalling. Instead, I flashed the SD card with Raspberian again then setup the seed-voicecard.

ismailryt, 2020-05-07:
@bencevans so you didnt do anything other? Im using raspy4 and latest buster, and i already try several time reflash SD card and do sudo ./install.sh , but arecord -l didnt detect any device. Like what you post 

 So if you have any advice tell me, thank you

amirush786, 2021-08-05:
> Duplicate of #166

I am working with 6 mic tool to collect data for a pose estimation project.
I am not able to record any sound from the 6 microphone tool. I tried fixing it but I don't know what's wrong. I have attached the expected and actual output when I am running. Can you please help me with it.  website for 6 mic tool 
https://wiki.seeedstudio.com/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/