respeaker/seeed-voicecard #170: Respeaker 2-Mics PI on Rasbperry Pi Zero does not recognize speaker and headset devices

Jesse0617, 2019-07-20:
Linux version: Linux raspberrypi 4.19.58+ #1245 Fri Jul 12 17:20:08 BST 2019 armv6l GNU/Linux

The issue seems to be almost same as #159 but it is not fixed.
I tried to downgrade karnel version to v4.14 and reboot and restart but I could not solve it.

The problem is here.
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~ $ 
```
Would you please help me think about next action ?

alenweiru, 2019-07-20:
i have the same issue ,the follow command worked for me 
```
sudo rpi-update ee41ed73e89e198aaa84e99bd53c352203d125f8
sudo reboot 
sudo ./install.sh
```

Jesse0617, 2019-07-20:
Yeah,  I did the same but it is not fixed yet.
```
pi@raspberrypi:~ $ uname -ar
Linux raspberrypi 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
```

alenweiru, 2019-07-21:
I think one thing you can check is that whether ReSpeaker module 40 pin  headers is fit to raspberry or not

Jesse0617, 2019-07-23:
I already checked the connection but its fine. 
[image]
[image]

Pillar1989, 2019-07-25:
We tested Rasbperry Pi Zero and found no problem you mentioned

closed 2019-08-20 by Pillar1989