respeaker/seeed-voicecard #174: make an install for alpine linux

vincentserpoul, 2019-08-20:
How difficult would it be to create an alpine version?

Pillar1989, 2019-08-20:
I don't understand your issue. ..

vincentserpoul, 2019-08-20:
this repo only works for raspbian distribs (apt-get ...)
How much work would it be to make it work for alpine linux?

Pillar1989, 2019-09-03:
I dont know

closed 2019-09-03 by Pillar1989

vincentserpoul, 2019-09-03:
at least, that's honest :)

biemster, 2020-11-15:
very honest indeed :D @vincentserpoul did you manage? I'm facing the same issue atm.

vincentserpoul, 2020-11-15:
unfortunately,  I gave up and ran a raspbian for the specific rpi running the speaker :/

biemster, 2020-11-15:
Arch seems to just have added this to the AUR:
https://aur.archlinux.org/packages/seeed-voicecard-dkms-git/
so I'm going to see if I can reproduce in Alpine what they did in Arch. I'll report back here when successful. @Pillar1989 can you reopen this issue?

biemster, 2020-11-15:
Compiling the kernel modules and ac108 lib goes without issues, after adding the linux-rpi4-dev apk. No need for dkms. The rest is figuring out where to copy everything (I have to figure out alsa now, so this could take a bit longer).
I'm actually running my pi via [netboot](https://github.com/biemster/rpi4_alpine_netboot), so my install will probably a be a bit unconventional.

HinTak, 2020-11-15:
The dkms side is easy. The harder part is where to put the devicetree overlays and, if your distribution does not ship the device tree overlay tools. Look at the repo for `git log --grep=ubuntu` for the recent ubuntu-related changes, and please feel free to donate at https://hintak.github.io ...

biemster, 2020-11-16:
Thanks for your input @HinTak . When I connected my tvhat to this rpi and booted it up, it just asked for the tv-hat dtbo over tftp (this rpi is netbooting). Do you have any clue how the tvhat module made that happen (I did not change anything in the config.txt, just connected the hat), and could this be built for the voicecard too?

HinTak, 2020-11-16:
I don't think that by "asked", you mean "prompted". I don't know how the tvhat works, but I believe the 40-pin GPIO header hardware does not support hot-plug, so you just mean that if the hat is present before your pi is powered up, its dtbo and driver is loaded automatically.

That I think can happen if there is a start-up process (not necessarily systemd based but probably is ),which scans the i2c bus for the possibility that a tvhat is present, and load the dtbo, which in turn loads the kernel modules.

The seeed studio driver does similar thing - there is systemd service which scans the i2c bus and see which of the mic arrays (there are 4, I think) may be present, and load the correct one, which then loads the kernel modules.

For net-boot, I think you need to put the systemd service script itself, the script it runs, the dtbo and the kernel modules onto the initramfs.

biemster, 2020-11-16:
> I don't think that by "asked", you mean "prompted".
I don't think that either :+1: 

Anyway, thanks for your detailed explanation. I'll post instructions in this thread if I get it to work.

biemster, 2020-11-18:
So this is how far I got up to now:
dtbo is loaded by adding the following to `config.txt`:
```
dtparam=i2c_arm=on,spi=on
dtoverlay=seeed-4mic-voicecard
```
this will load the i2c and spi modules as well (and added i2c-dev to the modules param in cmdline.txt):
```
Module                  Size  Used by    Tainted: G  
af_packet              57344  0 
spidev                 24576  0 
snd_soc_ac108          69632  0 
snd_soc_seeed_voicecard    16384  1 snd_soc_ac108
snd_soc_simple_card_utils    20480  1 snd_soc_seeed_voicecard
snd_soc_core          217088  3 snd_soc_ac108,snd_soc_seeed_voicecard,snd_soc_simple_card_utils
snd_pcm_dmaengine      20480  1 snd_soc_core
snd_pcm               126976  2 snd_soc_core,snd_pcm_dmaengine
snd_timer              45056  1 snd_pcm
snd_compress           20480  1 snd_soc_core
snd                   102400  4 snd_soc_core,snd_pcm,snd_timer,snd_compress
spi_bcm2835            24576  0 
i2c_bcm2835            16384  0 
i2c_dev                20480  0 
squashfs               53248  0 
```
The only module I'm missing compared to a working raspberry OS is regmap_i2c, but this is a built-in now according to the defconfig:
```
CONFIG_REGMAP_I2C=y
```

i2cdetect prints
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```
which is also the same as on a working rpi OS install.

I've added the correct (for my 4 mic array) alsa config files to `/etc/asound.conf` and `/var/lib/alsa/asound.state`, so as far as I can make out from the install and service files, this should be about it.

However, alsactl prompts me
```
alsactl: load_state:1735: No soundcards found...
```
and `alsamixer`
```
cannot open mixer: No such file or directory
```

I will investigate further, but I'm a bit out of ideas tbh.

biemster, 2020-11-18:
It looks like I'm missing the following lines in dmesg:
```
[   30.021868] ac10x-codec 1-003b: ac108_set_sysclk freq = 24000000 clk = 0
[   30.021886] ac108_set_sysclk  :24000000
[   30.031984] seeed-voicecard soc:sound: ac10x-codec.1-003b <-> 20203000.i2s mapping ok
```
which I took from a working rpi OS install with the array attached.

biemster, 2020-11-18:
So that was it! Just i2s was missing, it's working now.
To recap, the following is needed to get (at least the 4mic) array working on alpine:

1. Compile the kernel modules
`apk add linux-rpi-dev` (or `rpi4`)
run `make` (not `install.sh`)
2. copy the modules and the dbto to the correct location
3. add the following to `config.txt`
```
dtparam=i2c_arm=on,i2s=on,spi=on
dtoverlay=seeed-4mic-voicecard
```
4. copy the correct asound conf to `/etc/asound.conf` and state to `/var/lib/alsa/asound.state`
5. Now alsamixer should show the 4 mics!
6. start alsa deamon, also on startup
```
rc-service alsa start
rc-update add alsa
```

I can't really put this in an easy script because I netboot my machines. For me this means adding above kernel modules, and a few more sound related ones, to the initramfs. And providing the dtbo over TFTP.
Thanks @HinTak for your support!

vincentserpoul, 2020-11-18:
@biemster I'm impressed!
well done. I'll try it on my side

HinTak, 2020-11-18:
@biemster welldone indeed! One more linux distro works.

There is a donate link at the bottom of https://hintak.github.io . Buy me a beer if you wish!

biemster, 2020-11-19:
For the bonus round, below is a python snippet for the LEDs on the 4mic array. I could not get the spidev and gpiozero python modules running as given on the seeed website, so I picked gpiod. Don't forget to run the two commands (`apk add ...` and `pip install ...`) in a normal shell, and have SPI **switched off** on the rpi before running this:
```
#!/usr/bin/env python3
# apk add py3-pip py3-libgpiod
# pip install apa102-gpiod
import gpiod
import apa102_gpiod.apa102 as apa102
from time import sleep

nupdates = 100 # number of updates to the strip before shutting down
update_interval = 0.1 # seconds

# power on the strip by setting gpio 5 high
gpio5 = gpiod.Chip('/dev/gpiochip0').get_line(5)
gpio5.request(consumer='leds.py', type=gpiod.LINE_REQ_DIR_OUT, default_val=0)
gpio5.set_value(1)

# we have 12 LEDs on gpiochip0, connected to SPI0 MOSI (GPIO10) and CLK (GPIO11)
nleds = 12
pin_clk = 11
pin_data = 10
leds = apa102.APA102('/dev/gpiochip0', nleds, pin_clk, pin_data, True)

# set colors
colors = [(255,0,0),(0,255,0),(0,0,225)] * (nleds//3)
brightness = [1] * nleds # in range(32)
for u in range(nupdates):
    for led in range(nleds):
        leds[(led+u) % nleds] = apa102.LedOutput(brightness[led], *colors[led])
    leds.commit()
    sleep(update_interval)

# shut down
leds.close()
gpio5.set_value(0)
```

HinTak, 2020-11-23:
@biemster I wonder what sort of problems you had with the LEDs? I found the pixel_ring repo okay - yes, documentation is poor, but once you follow the installation instructions, and just blindly run the examples in the "examples" directory, some of them do something interesting and visible. There is a "respeaker 4 mics" example, which also works on the 6-mics array.

I have adapted /changed that to do a simple spinning rainbow (i.e. Just red/yellow/green/blue/purple etc and a few not-lit, rotating), mostly just to teach myself how to change colour .If you are interested I'll put it up somewhere.

I also have a different script which plays with the two on-board LEDs in the pi 4, basically make them blink rather than staying on as per usual.

biemster, 2020-11-23:
I did not have problems with the LEDs themselves, but installing the spidev and gpiozero python modules did not work out of the box on my minimal install. Those are not pure python, and require some C code to be build. At a certain point I needed some linux header files which didn't seem to have gotten installed when I built the kernel modules.
At that point I opted to go for pure python packages, also because they would be much easier to integrate in my lean setup.
Downside is however that they don't use the SPI interface, so they can't drive the LEDs as fast as the seeed provided solution.

muellerj, 2024-09-13:
Hi, I'm trying to get the HAT running unter Alpine and stumbled across this thread.

> 2\. copy the modules and the dbto to the correct location
Sorry if this a newbie question, but where do they need to be copied specifically?