respeaker/seeed-voicecard #175: kernel headers error, Echo error, Don't show in aplay, arecord, alsamixer

kajo222, 2019-08-20:
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh 6mic
Stary:1 http://archive.raspberrypi.org/debian stretch InRelease
Stary:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
All packages are up to date.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
raspberrypi-kernel is already the newest version (1.20190819~stretch-1).
raspberrypi-kernel-headers is already the newest version (1.20190819~stretch-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u4).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
Następujący pakiet został zainstalowany automatycznie i nie jest już więcej wymagany:
  realpath
Aby go usunąć należy użyć "sudo apt autoremove".
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.66+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

snd-soc-wm8960.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66+/
/usr/sbin/dkms: linia 1714: cd: /lib/modules/4.19.66+: Nie ma takiego pliku ani katalogu
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-ac108.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66+/
/usr/sbin/dkms: linia 1714: cd: /lib/modules/4.19.66+: Nie ma takiego pliku ani katalogu
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

snd-soc-seeed-voicecard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66+/
/usr/sbin/dkms: linia 1714: cd: /lib/modules/4.19.66+: Nie ma takiego pliku ani katalogu
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...(bad exit status: 1)

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  seeed-voicecard
Version: 0.3
Kernel:  4.19.66-v7+ (armv7l)
-------------------------------------

[... 238 lines cut ...]
create seeed-8mic-voicecard asound status file
+ ln -s /etc/voicecard/ac108_6mic.state /var/lib/alsa/asound.state
+ alsactl restore
Found hardware: "bcm2835_alsa" "Broadcom Mixer" "" "" ""
Hardware is initialized using a generic method
alsactl: set_control:1325: failed to obtain info for control #6 (No such file or directory)
+ amixer cset numid=3 1
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=3,step=0
  : values=1

pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- UU -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

I tried removing, reinstalling the kernel, modifying the install.sh file to manually write the kernel, Different versions of the system (even 2 years older), modify /boot/config.txt, restart the seeed service. I tested with apt-upgrade and without

Please Help me

Pillar1989, 2019-08-22:
I suggest you test the hardware and our software with an lastest system.

kajo222, 2019-08-22:
I do it and don't work

Pillar1989, 2019-08-22:
[image]
Did you delete it?

kajo222, 2019-08-22:
no

Pillar1989, 2019-08-22:
Show me. I don't believe it.

kajo222, 2019-08-22:
kacper@kacper-INVALID:~$ sudo ssh pi@10.10.0.59
pi@10.10.0.59's password: 
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Aug 22 10:06:27 2019

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ ls
4.19.66+      Desktop    Downloads  Music     Public           Templates
4.19.66-v7l+  Documents  MagPi      Pictures  seeed-voicecard  Videos
pi@raspberrypi:~ $ cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ ls /lib/mod
modprobe.d/ modules/    
pi@raspberrypi:~/seeed-voicecard $ ls /lib/modules/
4.19.66-v7+
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh 6mic
Stary:1 http://archive.raspberrypi.org/debian buster InRelease
Stary:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
All packages are up to date.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
raspberrypi-kernel is already the newest version (1.20190819-1).
Następujące pakiety zostały zainstalowane automatycznie i nie są już więcej wymagane:
  python3-pyperclip python3-thonny rpi.gpio-common
Aby je usunąć należy użyć "sudo apt autoremove".
Zostaną zainstalowane następujące NOWE pakiety:
  raspberrypi-kernel-headers
0 aktualizowanych, 1 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Konieczne pobranie 24,9 MB archiwów.
Po tej operacji zostanie dodatkowo użyte 163 MB miejsca na dysku.
Pobieranie:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20190819-1 [24,9 MB]
Pobieranie:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20190819-1 [24,9 MB]
Pobrano 3 207 kB w 19s (173 kB/s)                                              
Wybieranie wcześniej niewybranego pakietu raspberrypi-kernel-headers.
(Odczytywanie bazy danych ... 91923 pliki i katalogi obecnie zainstalowane.)
Przygotowywanie do rozpakowania pakietu .../raspberrypi-kernel-headers_1.20190819-1_armhf.deb ...
Rozpakowywanie pakietu raspberrypi-kernel-headers (1.20190819-1) ...
Konfigurowanie pakietu raspberrypi-kernel-headers (1.20190819-1) ...
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
git is already the newest version (1:2.20.1-2).
i2c-tools is already the newest version (4.1-1).
i2c-tools zaznaczony jako zainstalowany ręcznie.
Następujące pakiety zostały zainstalowane automatycznie i nie są już więcej wymagane:
  python3-pyperclip python3-thonny rpi.gpio-common
Aby je usunąć należy użyć "sudo apt autoremove".
Sugerowane pakiety:
  python3-apport menu
Zostaną zainstalowane następujące NOWE pakiety:
  dkms libasound2-plugins
0 aktualizowanych, 2 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
Konieczne pobranie 139 kB archiwów.
Po tej operacji zostanie dodatkowo użyte 565 kB miejsca na dysku.
Pobieranie:1 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf dkms all 2.6.1-4 [74,4 kB]
Pobieranie:2 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libasound2-plugins armhf 1.1.8-1 [64,2 kB]
Pobrano 139 kB w 1s (220 kB/s)           
Wybieranie wcześniej niewybranego pakietu dkms.
[... 168 lines cut ...]
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~ $ sudo service seeed-voicecard restart
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~ $

kajo222, 2019-08-22:
On the latest image i don't have that problem, but still don't work

kajo222, 2019-08-22:
Installed your preconfigured image and working, but i want know why before don't work

kajo222, 2019-08-22:
And after preconfig and raspi-config again don't work

kajo222, 2019-08-22:
i dont update btw

czw., 22 sie 2019 o 09:18 Kacper korczak <kacper.korczak123@gmail.com>
napisał(a):

kajo222, 2019-08-22:
I think it is 1-Wire problem, how i can use 1-wire and respeaker?

Pillar1989, 2019-09-03:
the  issue fixed?

kajo222, 2019-09-03:
YEAH

Pillar1989, 2019-09-03:
So let me close the issue, I hope you can come and communicate with us next time you encounter problems. I sincerely hope you are in good health and everything goes well.

closed 2019-09-03 by Pillar1989