respeaker/seeed-voicecard #177: Respeaker 4 Mic Array channel 3 missing

Mopele, 2019-08-25:
Hey, 
I wanted to use the 4 Mic array for Pi, but after the installation process described by the wiki I was not able to get all 4 channels in Audacity. I see all channels, but the 3. channel is just a empty signal. I also noticed, that on the input devies in audacity there are more options as only seeed-4mic-voicecard and ac108, there is also sysdefauld,lavrate,samplerate, spexrate, upmix and vdownmix.
If you need more information just ask.
Thanks for the help!

Pillar1989, 2019-09-03:
This may be a hardware problem. Please contact your supplier of cars and light trucks

closed 2019-10-11 by Pillar1989

HinTak, 2020-08-30:
@Mopele - this is probably the same issue as #240 - the seeedvoice card systemd service assumes that the on-board bcm driver is loaded, and set playback audio to the on-board phone jack with a amixer command. Unforturnately, if you don't use the pi's on-board audio (or with a later kernel where they are separate), the amixer command when applied to the wrong device, has the effect of turning the recording volume of the 3rd channel really low!