respeaker/seeed-voicecard #180: Not working on 4.19.71+ - Respeaker 2 and 6 mic array - RPi 3 and RPi 3B+

vedranius, 2019-09-14:
I'm getting a problem for kernel 4.19.71+ for Respeaker 2 and 6 mic array - RPi 3 and RPi 3B+

Can you fix it?!

 !!! Your kernel version is 4.19.71+
     Not found *** coressponding *** kernel headers with apt-get.
     This may occur if you have ran 'rpi-update'.
 Choose  *** y *** will revert the kernel to version 4.19.66+ then continue.
 Choose  *** N *** will exit without this driver support, by default.

Pillar1989, 2019-09-23:
@vedranius  Problem solved? Next time you ask a question remember to describe clearly, otherwise it will be easy to ignore.

Nokius, 2019-10-09:
@Pillar1989 it's the same issue as in #183 The newer kernel is not supported.

HinTak, 2019-10-14:
if you do apt upgrade now, it should all work. See https://github.com/raspberrypi/linux/issues/3279

turmary, 2019-10-23:
Will close this issue, please open a new issue if you have further questions.

closed 2019-10-23 by turmary