respeaker/seeed-voicecard #181: Problem installing

dksim, 2019-09-23:
Cannot get it ReSpeaker 4 Mic Array to work as after installing on `Linux raspberrypi 4.19.66-v7l+`
`arecord -L` gives

```
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
playback
ac108
```
`

Pillar1989, 2019-09-26:
It looks like the hardware raspberry PI didn't recognize the i2c address. Is your hat stuck on the board all the time?

dksim, 2019-09-26:
Yes, hat is attached to PI. Excerpt from install.sh

```
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.19.66+ -C /lib/modules/4.19.66+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.66+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66+/kernel/sound/soc/codecs/snd-soc-ac108.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.66+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66+/kernel/sound/soc/bcm/snd-soc-seeed-voicecard.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.66+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66+/kernel/sound/soc/bcm/

depmod....

DKMS: install completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.66-v7+ -C /lib/modules/4.19.66-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.66-v7+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/snd-soc-ac108.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.66-v7+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/snd-soc-seeed-voicecard.ko
   - Storing in /var/lib/dkms/seeed-voicecard/original_module/4.19.66-v7+/armv7l/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/kernel/sound/soc/bcm/

depmod....

DKMS: install completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.66-v7l+ -C /lib/modules/4.19.66-v7l+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7l+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7l+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7l+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) c2d424f] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.

```
`

dksim, 2019-10-04:
Needed clean Raspbian. However it only works up to 4.19.66.

closed 2019-10-04 by dksim