respeaker/seeed-voicecard #183: Respeaker 4-Mic-Array - Raspberry 3B+ - 4.19.75-v7+- Installation failed

WerrnerHamannPrivate, 2019-10-02:
**Installation**

I installed the new Raspian-Buster image (2019-09-26-raspbian-buster-full.img).

Then I took the following steps:

```
sudo apt-get update
sudo apt-get upgrade
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh

```
After this I get:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
Building dependency tree...
Reading state information...
All packages are up to date.
Reading package lists...
Building dependency tree...
Reading state information...
raspberrypi-kernel-headers is already the newest version (1.20190819-1).
raspberrypi-kernel is already the newest version (1.20190925-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 !!! Your kernel version is 4.19.75+
     Not found *** coressponding *** kernel headers with apt-get.
     This may occur if you have ran 'rpi-update'.
 Choose  *** y *** will revert the kernel to version 4.19.66+ then continue.
 Choose  *** N *** will exit without this driver support, by default.

Reading package lists...
Building dependency tree...
Reading state information...
Reinstallation of raspberrypi-kernel is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

------------------------------
Deleting module version: 0.3
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master
nothing to commit, working tree clean
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

That's all I did.
So it is a clean installation.

***
**arecord**

```
pi@raspberrypi:~/seeed-voicecard $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
playback
ac108
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output
pi@raspberrypi:~/seeed-voicecard $
```

*** uname ***
```
pi@raspberrypi:~/seeed-voicecard $ uname -r
4.19.75-v7+

```

wlmqpsc, 2019-10-03:
Please add the support of new kernel(4.19.75+), I have the same problem on 3b

Marv339, 2019-10-03:
I have the same problem on 4b

akira783, 2019-10-03:
Hi there !
Just got my RPi4 (with last Debian Buster) and Respeaker 4-Mic... Exactly the same issue :/

wlmqpsc, 2019-10-04:
It only support the Raspbian kernel 4.19.66+and lower.
we can download the raspbian history image at [here](http://downloads.raspberrypi.org/raspbian/images/)
and the newest supported version is [raspbian-2019-7-12](http://downloads.raspberrypi.org/raspbian/images/raspbian-2019-07-12/)

SargonofAssyria, 2019-10-05:
So I went back to raspbian-2019-7-12. That did not work, it asked for version 4.19.66
So I upgraded to 4.19.66  I want to install the Google Hat.
Now I am stuck with the install of 'sudo pip3 install tensorflow' It gives an error.
Anybody any idea?

SargonofAssyria, 2019-10-06:
Found it had to use --no-cache-dir
And I put updates for kernel etc.. on hold with:
sudo apt-mark hold raspberrypi-kernel
sudo apt-mark hold raspberrypi-bootloader
sudo apt-mark hold raspberrypi-kernel-headers
Now I am on 4.19.66-v7+ and everything is working fine.

HinTak, 2019-10-11:
new header package is out, see https://github.com/respeaker/seeed-voicecard/issues/190

juancarlocc, 2019-10-11:
@SargonofAssyria  How do I upgarde to 4.19.66-v7+? I'm currently on version 4.19.57-v7+. Also How do you use the --no-cache-dir?

HinTak, 2019-10-11:
@juancarlocc you can force install the kernel and headers to specific versions with:

```
sudo apt install raspberrypi-kernel=1.20190819-1 raspberrypi-kernel-headers=1.20190819-1
```

I had this working, before I experimented with the latest, and if you follow the instruction in https://github.com/respeaker/seeed-voicecard/issues/190 , you can use 1.20190925-2 / 4.19.75-v7+ too. Either way works ; the latter is slightly future-proof.

SargonofAssyria, 2019-10-11:
Or get to a specific kernel:
`sudo rpi-update <commit-id>`
You can find those at: https://github.com/Hexxeh/rpi-firmware/commits/master
More info: https://github.com/Hexxeh/rpi-update

wlmqpsc, 2019-10-11:
I just use 2019-7-12 and use `sudo apt-get update && sudo apt-get upgrade`.It changed to the 4.19.66-v7+

HinTak, 2019-10-11:
1.20190925-2 / 4.19.75-v7+  was available within the last 24 hours, and you'd need to read https://github.com/respeaker/seeed-voicecard/issues/190 if you just do plain `apt-get update ...` .

juancarlocc, 2019-10-11:
@HinTak thanks mine worked now I followed the link you shared. But now there is a new issue whenever I run alexa

ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM input
arecord: main:828: audio open error: No such file or directory

HinTak, 2019-10-11:
That looks more like an alsa config issue. Did you have it working in a previous raspbian release?

juancarlocc, 2019-10-11:
Ok I don't know why this worked but it worked:
$ sudo pip install voice_engine

HinTak, 2019-10-14:
If you do at upgrade now, it should all work. See https://github.com/raspberrypi/linux/issues/3279

turmary, 2019-10-23:
Will close this issue, please open a new issue if you have further questions.

closed 2019-10-23 by turmary