respeaker/seeed-voicecard #184: Unable to use the codec with Jetson TX2

AshaTalambedu, 2019-10-04:
I am trying to integrate Re-Speaker 4-Mic Array to Jetson TX2 which comes with Linux BSP. I see that DAPM path is not getting setup properly. In DT overlay provided, there is no routing field in the sound node. I understand that MIC1N, MIC1P etc.. are the physical codec pins that receive analog data on codec from the available microphones. 

Therefore, I have added below routes in sound node for the  DAI link whose name prefix is "x"

                      "x MIC1N",              "x Mic",
                      "x MIC1P",              "x Mic",
                      "x MIC2N",              "x Mic",
                      "x MIC2P",              "x Mic",
                      "x MIC3N",              "x Mic",
                      "x MIC3P",              "x Mic",
                      "x MIC4N",              "x Mic",
                      "x MIC4P",              "x Mic",
The DAPM path is broken as there is no sink for MIC1N, MIC1P so on. on the codec side.

Also, I see below widget
SND_SOC_DAPM_AIF_IN("Channel 4 AAF", "Capture", 0, ANA_ADC4_CTRL1, ADC4_DSM_ENABLE, 1),

In other codec drivers, I see that capture stream is associated with AIF_OUT widget in general as it transmits data to I2S Receiver of the host, essentially making it an output. Why is this different here in ac108 codec. Also, I am not sure how is this working on RaspPi with these issues

Can anyone help rectify the DAPM path broken issue and help me in clarifying the doubts

AshaTalambedu, 2019-10-04:
Note that i am trying to run in codec slave mode

Pillar1989, 2019-10-24:
Jetson TX2 , This supports four channel voice?

Pillar1989, 2019-10-24:
@AshaTalambedu

AshaTalambedu, 2019-12-27:
I am using the host I2S in TDM mode

AshaTalambedu, 2019-12-27:
I have got ac108 driver from x-power. With the driver, I am able to resolve DAPM path issue. I have configured the SYSCLK to be derived from PLL which in turn is derived from SoC I2S BCLK. But silence is getting captured. Any idea, what could have gone wrong?

Pillar1989, 2019-12-27:
So maybe there's a problem with the dapm path.

Pillar1989, 2019-12-27:
If we use our driver completely and use BCLK, there should be no DAPM path problems.
This is the right DAPM.
https://github.com/respeaker/seeed-voicecard/blob/master/ac108_asound.state

AshaTalambedu, 2019-12-27:
Does this driver work on Linux platforms?

Pillar1989, 2019-12-27:
Of course

AshaTalambedu, 2019-12-27:
can you paste the DAPM trace from your RPi using below commands while running capture usecase

for i in `find /sys/kernel/debug/tracing/events -name "enable" | grep snd_soc_`; do echo 1 > $i; done
cat /sys/kernel/debug/tracing/trace_pipe

Pillar1989, 2019-12-27:
I don't have this product at present, it has been sent to the warehouse.If I can find one handy today, I'll print it out for you

AshaTalambedu, 2019-12-27:
Sure thanks!

With Jetson TX2 following are the setting I am using and dapm path trace is as shown below

amixer -c 1 cset name="x ADC1 DIG MIXER ADC1 DAT switch" 1
amixer -c 1 cset name="x ADC2 DIG MIXER ADC2 DAT switch" 1
amixer -c 1 cset name="x ADC3 DIG MIXER ADC3 DAT switch" 1
amixer -c 1 cset name="x ADC4 DIG MIXER ADC4 DAT switch" 1

amixer -c 1 cset name="x ADC1 DIG MUX" "ADC1 switch"
amixer -c 1 cset name="x ADC2 DIG MUX" "ADC2 switch"
amixer -c 1 cset name="x ADC3 DIG MUX" "ADC3 switch"
amixer -c 1 cset name="x ADC4 DIG MUX" "ADC4 switch"

amixer -c 1 cset name="x I2S TX1 CH1 MUX" "ADC1 Sample switch"
amixer -c 1 cset name="x I2S TX1 CH2 MUX" "ADC2 Sample switch"
amixer -c 1 cset name="x I2S TX1 CH3 MUX" "ADC3 Sample switch"
amixer -c 1 cset name="x I2S TX1 CH4 MUX" "ADC4 Sample switch"

amixer -c 1 cset name="x ADC12 DMIC1 MUX" "ADC12 switch"
amixer -c 1 cset name="x ADC34 DMIC2 MUX" "ADC34 switch"

amixer -c 1 cset name="x ADC1 PGA gain" 15
amixer -c 1 cset name="x ADC2 PGA gain" 15
amixer -c 1 cset name="x ADC3 PGA gain" 15
amixer -c 1 cset name="x ADC4 PGA gain" 15

amixer -c 1 cset name='x CH1 digital volume' 120
amixer -c 1 cset name='x CH2 digital volume' 120
amixer -c 1 cset name='x CH3 digital volume' 120
amixer -c 1 cset name='x CH4 digital volume' 120

amixer -c 1 cset name="x CH1 ch1 mixer gain" 1
amixer -c 1 cset name="x CH2 ch2 mixer gain" 1
amixer -c 1 cset name="x CH3 ch3 mixer gain" 1
amixer -c 1 cset name="x CH4 ch4 mixer gain" 1

amixer -c 1 cset name="ADMAIF1 Mux" I2S1
amixer -c 1 cset name="I2S1 Mux" 0

arecord -D hw:1,0 -f S32_LE -c 4 -r 48000 in.wav &

DAPM trace:
         arecord-5121  [002] ....  1758.861225: snd_soc_dapm_path: *ADMAIF1 TX -> (direct) -> ADMAIF1 Transmit
         arecord-5121  [002] ....  1758.861243: snd_soc_dapm_path: *ADMAIF1 Mux -> (direct) -> ADMAIF1 TX
         arecord-5121  [002] ....  1758.861361: snd_soc_dapm_path: *I2S1 RX -> I2S1 -> ADMAIF1 Mux
         arecord-5121  [002] ....  1758.861365: snd_soc_dapm_path: *I2S1 Receive -> (direct) -> I2S1 RX
         arecord-5121  [002] ....  1758.861367: snd_soc_dapm_path: *I2S1 CIF Transmit-I2S1 Receive -> (direct) -> I2S1 Receive
         arecord-5121  [002] ....  1758.861371: snd_soc_dapm_path: *I2S1 CIF Transmit -> (direct) -> I2S1 CIF Transmit-I2S1 Receive
         arecord-5121  [002] ....  1758.861374: snd_soc_dapm_path: *I2S1 CIF TX -> (direct) -> I2S1 CIF Transmit
         arecord-5121  [002] ....  1758.861377: snd_soc_dapm_path: *I2S1 DAP RX -> (direct) -> I2S1 CIF TX
         arecord-5121  [002] ....  1758.861380: snd_soc_dapm_path: *I2S1 DAP Receive -> (direct) -> I2S1 DAP RX
         arecord-5121  [002] ....  1758.861383: snd_soc_dapm_path: *x Capture-I2S1 DAP Receive -> (direct) -> I2S1 DAP Receive
         arecord-5121  [002] ....  1758.861386: snd_soc_dapm_path: *x Capture -> (direct) -> x Capture-I2S1 DAP Receive
         arecord-5121  [002] ....  1758.861390: snd_soc_dapm_path: *x AIF ADC OUT -> (direct) -> x Capture
         arecord-5121  [002] ....  1758.861393: snd_soc_dapm_path: *x I2S TX1 CH4 MUX -> (direct) -> x AIF ADC OUT
         arecord-5121  [002] ....  1758.861396: snd_soc_dapm_path: *x I2S TX1 CH3 MUX -> (direct) -> x AIF ADC OUT
         arecord-5121  [002] ....  1758.861399: snd_soc_dapm_path: *x I2S TX1 CH2 MUX -> (direct) -> x AIF ADC OUT
         arecord-5121  [002] ....  1758.861417: snd_soc_dapm_path: *x I2S TX1 CH1 MUX -> (direct) -> x AIF ADC OUT
         arecord-5121  [002] ....  1758.861420: snd_soc_dapm_path: *x ADC4 DIG MIXER -> ADC4 Sample switch -> x I2S TX1 CH4 MUX
         arecord-5121  [002] ....  1758.861428: snd_soc_dapm_path: *x ADC3 DIG MIXER -> ADC3 Sample switch -> x I2S TX1 CH3 MUX
         arecord-5121  [002] ....  1758.861437: snd_soc_dapm_path: *x ADC2 DIG MIXER -> ADC2 Sample switch -> x I2S TX1 CH2 MUX
         arecord-5121  [002] ....  1758.861446: snd_soc_dapm_path: *x ADC1 DIG MIXER -> ADC1 Sample switch -> x I2S TX1 CH1 MUX
         arecord-5121  [002] ....  1758.861450: snd_soc_dapm_path: *x ADC4 VIR PGA -> ADC4 DAT switch -> x ADC4 DIG MIXER
         arecord-5121  [002] ....  1758.861459: snd_soc_dapm_path: *x ADC3 VIR PGA -> ADC3 DAT switch -> x ADC3 DIG MIXER
         arecord-5121  [002] ....  1758.861467: snd_soc_dapm_path: *x ADC2 VIR PGA -> ADC2 DAT switch -> x ADC2 DIG MIXER
         arecord-5121  [002] ....  1758.861506: snd_soc_dapm_path: *x ADC1 VIR PGA -> ADC1 DAT switch -> x ADC1 DIG MIXER
         arecord-5121  [002] ....  1758.861509: snd_soc_dapm_path: *x ADC34 DMIC2 MUX -> (direct) -> x ADC4 VIR PGA
         arecord-5121  [002] ....  1758.861510: snd_soc_dapm_path: *x ADC34 DMIC2 MUX -> (direct) -> x ADC3 VIR PGA
         arecord-5121  [002] ....  1758.861513: snd_soc_dapm_path: *x ADC12 DMIC1 MUX -> (direct) -> x ADC2 VIR PGA
         arecord-5121  [002] ....  1758.861514: snd_soc_dapm_path: *x ADC12 DMIC1 MUX -> (direct) -> x ADC1 VIR PGA
         arecord-5121  [002] ....  1758.861517: snd_soc_dapm_path: *x ADC4 DIG MUX -> ADC34 switch -> x ADC34 DMIC2 MUX
         arecord-5121  [002] ....  1758.861519: snd_soc_dapm_path: *x ADC3 DIG MUX -> ADC34 switch -> x ADC34 DMIC2 MUX
         arecord-5121  [002] ....  1758.861522: snd_soc_dapm_path: *x ADC2 DIG MUX -> ADC12 switch -> x ADC12 DMIC1 MUX
         arecord-5121  [002] ....  1758.861524: snd_soc_dapm_path: *x ADC1 DIG MUX -> ADC12 switch -> x ADC12 DMIC1 MUX
[... cut ...]
         arecord-5121  [002] ....  1758.861574: snd_soc_dapm_path: *x Mic -> (direct) -> x MIC3P
         arecord-5121  [002] ....  1758.861575: snd_soc_dapm_path: *x Mic -> (direct) -> x MIC3N
         arecord-5121  [002] ....  1758.861576: snd_soc_dapm_path: *x Mic -> (direct) -> x MIC2P
         arecord-5121  [002] ....  1758.861577: snd_soc_dapm_path: *x Mic -> (direct) -> x MIC2N
         arecord-5121  [002] ....  1758.861578: snd_soc_dapm_path: *x Mic -> (direct) -> x MIC1P
         arecord-5121  [002] ....  1758.861579: snd_soc_dapm_path: *x Mic -> (direct) -> x MIC1N

AshaTalambedu, 2020-03-02:
Hi,
I have another query. Will respeaker drivers be upstreamed on Linux Kernel?
Is there a plan in this regard?

Thanks

closed 2020-03-03 by AshaTalambedu