respeaker/seeed-voicecard #186: respeaker 2 mic hat installation failing - Error Error! There are no instances of module: seeed-voicecard 0.3 located in the DKMS tree.

baburaaj, 2019-10-09:
I was trying to install the driver for the respeaker 2 mic hat . I got the error above . the following is what I did
sudo ./install.sh
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20190819-1).
raspberrypi-kernel is already the newest version (1.20190925-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 !!! Your kernel version is 4.19.75+
     Not found *** coressponding *** kernel headers with apt-get.
     This may occur if you have ran 'rpi-update'.
 Choose  *** y *** will revert the kernel to version 4.19.66+ then continue.
 Choose  *** N *** will exit without this driver support, by default.
Would you like to proceed? (y/N)
Reading package lists... Done
Building dependency tree
Reading state information... Done
Reinstallation of raspberrypi-kernel is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.75+ cannot be found at
/lib/modules/4.19.75+/build or /lib/modules/4.19.75+/source.
Error! echo
Your kernel headers for kernel 4.19.75-v7+ cannot be found at
/lib/modules/4.19.75-v7+/build or /lib/modules/4.19.75-v7+/source.
Error! echo
Your kernel headers for kernel 4.19.75-v7l+ cannot be found at
/lib/modules/4.19.75-v7l+/build or /lib/modules/4.19.75-v7l+/source.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) db615f1] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 [asound_4mic.conf](url)
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service                                                                                                              → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
[(https://github.com/respeaker/seeed-voicecard/issues/100)] . I had tried everything here. I had removed all the older kernel folders . but still the issue persists  and it still shows the list of kernels . I dont know what to do now . I am new to this and would require some hand holding . 
Please help

Nokius, 2019-10-09:
@baburaaj try the solution with the [apt pinning mentioned](https://github.com/respeaker/seeed-voicecard/issues/183#issuecomment-538719971) in #183

HinTak, 2019-10-11:
same issue as https://github.com/respeaker/seeed-voicecard/issues/190

HinTak, 2019-10-14:
If you do apt upgrade now, it should work . See https://github.com/raspberrypi/linux/issues/3279

turmary, 2019-10-23:
Will close this issue, please open a new issue if you have further questions.

closed 2019-10-23 by turmary